adafruit/Adafruit_Python_DHT #9: Adafruit_Python3_Raspberry_Pi

bob60, 2015-01-03:


closed 2015-01-03 by bob60

tdicola, 2015-01-03:
Oh wow, nice!  If it's not too radically different we can merge it in to better support Python 3 in this library.  It's been on my todo list for a while, but if you've already knocked it out that's awesome. :)

bob60, 2015-01-04:
Hi Tony,

i’have some problem:

1)      English language

2)      Know-how on Github

3)      Not many time for me

 

I think the second and third can be resolved… I’m italian and my english is very… let it be!

I have modified the software relative to Pi because i not own a BB and then i can’t test it.

I’m learning about Github but if you want (and i prefere) i can send to you the entire directory modified so you can test it and maybe get the modify on Rasp and replicate on BB. The most important modify interest the driver and it can be replicated on BB driver.

 

Waiting for news from you 

Roberto 

A: adafruit/Adafruit_Python_DHT
Cc: bob60
Oggetto: Re: [Adafruit_Python_DHT] Adafruit_Python3_Raspberry_Pi (#9)