adafruit/Adafruit_Python_DHT #11: Update count logic to auto calibrate based on 80us pulse at start.

tdicola, 2015-01-17:
From mail, a user suggested changing the logic to set the count threshold based on the length of the first 80us pulse sent by the sensor.  From Bill's mail:

"The initial pulse from the device is 80µS according to the documentation and I think that is there to allow a controller to measure the time.

My change looks for "_count == -1" to indicate that it should self calibrate, so using something like "DHT dht(DHTPIN, DHTTYPE, -1);" to have it behave like this.

Here is the code to replace lines 147 to 154:

// ignore first 1 transition
if ( (i > 0) && (i%2 == 0)) {
  // Is it the initial 80µS pulse?
  if(i==2) {
    // Should automatic calibration be used?
    if(_count == -1) {
       // 1st pulse is 80µS long, so the "1" pulse of 70µS
       // should always be bigger than 1/2 of the counter
       _count = counter / 2;
    )
  }
  else {
    // shove each bit into the storage bytes
    data[j/8] <<= 1;
    if (counter > _count)
      data[j/8] |= 1;
    j++;
  }
}
"

Let's look at testing this change out and consider updating all the DHT libraries (Arduino, etc.) to use this kind of auto calibrating logic.

mvadu, 2015-04-20:
I implemented a different logic for AutoCalibrate using the 80uS pulses at the beginning. It certainly works (on my Pi2, with DHT11), and code becomes lot cleaner. 

What I noticed is that the first 80uS is always records a larger counter than the second one, and it makes sense because I am doing bit of work when I detect a state change. So I took half of the average of the two 80uS pulses as the 40uS value. That is compared against the bit values.

https://github.com/AdysTech/raspy-weather/blob/master/dht11_driver/pi_2_dht_read.c

ladyada, 2019-11-08:
hi this repo is going to be deprecated soon, please try 
https://github.com/adafruit/Adafruit_CircuitPython_DHT
which will replace this library

closed 2019-11-08 by ladyada