adafruit/Adafruit_Python_DHT #13: Python 3 support

tipa, 2015-03-01:
Is Python 3 support planned or already on its way?

NucleaPeon, 2015-05-25:
Python3 support would be greatly appreciated.

iorbitearth, 2015-11-11:
This module is working with python3 on the beaglebone black running Arch Linux. Tested on DHT 11.

M3taKn1ght, 2015-11-14:
I also tested it with a Pi an Python 3. There were no problems. You only have to include the lib and start with work

Cabalist, 2015-12-09:
@M3taKn1ght Could you explain what you mean by "include the lib and start with work"?  I am getting the following error on python3 after running ```sudo python setup.py install```

```
gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -I/usr/include/python2.7 -c source/_Raspberry_Pi_Driver.c -o build/temp.linux-armv6l-2.7/source/_Raspberry_Pi_Driver.o -std=gnu99
source/_Raspberry_Pi_Driver.c:21:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: command 'gcc' failed with exit status 1
```

bennuttall, 2015-12-09:
Try installing `python-dev`:

```bash
sudo apt-get install python-dev
```

Cabalist, 2015-12-09:
That did it.  I had ```python3-dev``` but apparently that is not base package I needed.  Thanks!

bychkov, 2016-02-03:
For those who need support for python3, you can find it in the following repo:
https://github.com/JoBergs/Adafruit_Python_DHT

ladyada, 2016-02-03:
@tdicola is spending this week and next doing a major overhaul of all our python libs to do: py3 support, pep8, and pypification. its...exciting and fun :) will close this issue once thats compete.

bennuttall, 2016-02-03:
<3

ladyada, 2016-02-03:
yep both python <3 and <2 will be supported ;)

bennuttall, 2016-02-03:
Haha

Djiit, 2016-05-22:
Hi there, 

I just tested the lib with Python3. Working like a charm!

Should we update the README and close this issue ?

tdicola, 2016-05-23:
Yep this one is all ready for Python 2 and 3 now.  Grab the latest code and it should install with either 2.7 or 3.3/3.4+.

closed 2016-05-23 by tdicola

bennuttall, 2016-05-23:
Great. Is there a way you can also remove the necessity for using sudo?

tomber42, 2017-06-30:
i want to install on python3 with sudo pip3 install adafruit_python_dht and got "ImportError: No module named 'platform_detect'"

Another try with sudo pip install adafruit_python_dht works, but module is only available in python2.7

I'm totaly new to python.. what i'm doing wrong?

tdicola, 2017-06-30:
You need to install from source like the guide mentions: https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/software-install-updated  The tricky thing about pip and python's package index is that it doesn't handle well a case where you have one platform (Linux) but different hardware-specific binaries.  In particular the library needs to compile special code for the Beaglebone Black vs. Raspberry Pi to do fast GPIO access.  Unfortunately folks have put the library on pip and it's really unknown what binary might be there--Beaglebone Black vs. Raspberry Pi vs. something else.  Stick to the setup.py install and it will compile and install it correctly for your platform.

jarvis-owl, 2018-03-04:
I got trapped with pip/pip3 and python/python3.
@tdicola helped me!

> sudo python**3** setup.py install