adafruit/Adafruit_Python_DHT #15: DHT22 bug

Dmole, 2015-03-04:
Sometimes the DHT22 will crash and will never pull the pin low thereafter until it has been power cycled.
Please update the install instructions to use a GPIO instead of the 3v rail.
Please add logic to automatically power cycle the DHT22 if it fails to pull the pin low.

(for testing I made DHT_MAXCOUNT 10 times bigger and made DHT_ERROR_TIMEOUT_2 and 3 to distinguish where the timeout was happening)

mvadu, 2015-04-20:
Thanks to your suggestion, I implemented a Power On when needed via GPIO logic in my version.

https://github.com/AdysTech/raspy-weather/blob/master/dht11_driver/dht11.c

    #define POWER_PIN 23
    #define SIGNAL_PIN 24
    
    
    void power_on_dht11()
    {
    	pi_2_mmio_set_output(POWER_PIN);
    	pi_2_mmio_set_high(POWER_PIN);
    
    	pi_2_mmio_set_input(SIGNAL_PIN);
    
    	//sleep_milliseconds(200);
    	volatile int t = 0;
    	//wait for GPIO to change and DHT11 to power on. Signal pin on DHT11 should have a pullup, so on powerup, it should be high
    	while (!pi_2_mmio_input(SIGNAL_PIN))
    	{
    		t++;
    		if (t > 200) break;
    	}
    	if (t > 200)		
    	{
    		printf("DHT11 was not detected!! \nMake sure the sensor is connected and there is a 4.7k pullup between +ve and signal pins.\n");
    		pi_2_mmio_set_low(POWER_PIN);
    		exit(-1);
    	}
    	return;
    }

Mausy5043, 2016-01-26:
@Dmole : I have not (yet) seen this happen. What symptoms should I be looking for? Can I detect this using Python?

I'm considering deploying a DHT22 connected to a BeagleBone Black remotely. So it would be nice if I could write some code that would detect the failure and power-cycle the DHT.

@mvadu :  :+1: it would be nice if you could make that into some generic code and offer a PR for @adafruit to implement.

Dmole, 2016-01-26:
@Mausy5043 Somewhere there is a kernel driver that should be used instead. The Adafruit code did not expose enough from c to do it in python, and python is to slow to avoid using c. The symptom was noted in the pull request with DHT_ERROR_TIMEOUT_2 (as if the device was unplugged).

Mausy5043, 2016-01-26:
So, if I understand correctly one could "just" test [after this](https://github.com/adafruit/Adafruit_Python_DHT/blob/master/Adafruit_DHT/Beaglebone_Black.py#L208) for `result == DHT_ERROR_TIMEOUT` coming from [here](https://github.com/adafruit/Adafruit_Python_DHT/blob/master/source/Beaglebone_Black/bbb_dht_read.c#L80) and then powercycle the sensor using @mvadu's code. RIght?

I have several more sensors on the 3V3 and 5V rails so, I'll need to time the powercycle just right not to disturb the other measurements. :wink:

Dmole, 2016-01-26:
@Mausy5043 , no because the c is not kernel code it's prone to timing out and erroring out regularly, so you need know where in the c code the timeout happened to know if you should just try again or if you need to power cycle it. powercycling the sensor is trivial and can be done in any language. It's so unstable (and hacky with the magic number) the example py code tries 3 times to make calls more reliable.

tdicola, 2016-06-22:
Hrm I haven't seen the need to power cycle sensors with the testing I've done.  For now I think we want to keep the examples simple and just power from 3v.  Thanks for the suggestion!

closed 2016-06-22 by tdicola