adafruit/Adafruit_Python_DHT #20: DHT11 on Jessie errors on GPIO on Raspberry Pi 2

klattimer, 2015-06-14:
root@jessie-rpi:~/Adafruit_Python_DHT/examples# python AdafruitDHT.py 11 4
Traceback (most recent call last):
  File "AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 90, in read_retry
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 77, in read
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/Raspberry_Pi_2.py", line 34, in read
RuntimeError: Error accessing GPIO. Make sure program is run as root with sudo!

I'm currently running as root, I also tried setting the gpio_base manually to 0x37200000 which is the correct address on Raspberry Pi 2 for the GPIO

BernardoSilva, 2015-06-16:
Same issue for me.

tdicola, 2015-06-18:
Huh interesting, right now code uses the same logic as the RPi.GPIO library to find the gpio base address from the device tree configuration.  Perhaps something is different here in Jessie and it's getting a garbage value (although if you manually set the GPIO and still had problems then that's not it).  I don't have a card upgraded to jessie handy to try with just yet, but next time I do I'll try to run through the code here and see if I can investigate further.

Also the error that it needs to run as root is thrown when it can't access /dev/mem or has a problem setting up the memory mapping GPIO (in this function: https://github.com/adafruit/Adafruit_Python_DHT/blob/master/source/Raspberry_Pi_2/pi_2_mmio.c#L38).  Likely something there in the memory mapping setup is behaving differently on Jessie, but I'm not sure off-hand what might be different with Jessie's kernel, updated libraries, etc.  Will need to dig in and check exactly what is failing.

johncstone, 2015-08-04:
Having exactly the same problems with the same routines - messages are the same as above.  Also using pi 2 with DHT11.

klattimer, 2015-08-04:
I switched to using WiringPI for the DHT11 because this code is basically useless.

gutopa, 2015-08-06:
Hello.

I'm also having this issue. Did anyone find a fix?

klattimer, 2015-08-06:
Switch to using wiring PI as it actually works, this code base is basically useless. (Second time I've posted this comment, but it looks like it has been removed, dear project owner, if you don't like your code being criticised then fix it when people want to use it instead of censoring criticism).

Meanwhile, drop me a message if anyone wants to take a copy of my DHT11 hacks for WiringPi but they're pretty much just a little ascii adjustments around the stock WiringPi example code.

johncstone, 2015-08-06:
Thanks for reply.  I too found WiringPi working better, but values are off a lot (temperature reading 512 degrees C).  I would be very appreciative to see what you did.

tdicola, 2015-08-07:
I just did a dist upgrade on a Raspbian wheezy card to take it to jessie and unfortunately I can't repro the same error.  Here's a quick example showing jessie and it working:
```
pi@raspberrypi ~/Adafruit_Python_DHT/examples $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
pi@raspberrypi ~/Adafruit_Python_DHT/examples $ sudo ./AdafruitDHT.py 2302 4
Temp=22.8*C  Humidity=77.9%
```
For folks that are running into problems I'm curious how did you upgrade to jessie?  I took a wheezy image (the 2015-05-05 image on raspberrypi.org right now), edited the /etc/apt/sources.list to reference jessie, did a apt-get update, then did a apt-get dist-upgrade.  The dist upgrade took a long time (about 7 hours, ouch!) but when it finished I installed the DHT python code and it seems to work like above.

One thing I have a suspicion about, did you use the image from raspbian.org?  I don't think you can really use that image since it's not from the Pi foundation and doesn't include their changes.  There's some stuff like reading the offset of the GPIO registers from device tree config that the DHT code depends on and might be missing from the raspbian.org image.  Try doing a dist upgrade on a Pi foundation wheezy image instead.

closed 2015-08-07 by tdicola

klattimer, 2015-08-07:
@tdicola censoring my comments further makes you look more and more like you can't stand basic criticism. Another one deleted.

ladyada, 2015-08-07:
@klattimer Please be positive and constructive with your questions and comments, this is your only warning, next time we'll block/delete, thank you!

klattimer, 2015-08-07:
Delete me from github? That doesn't sound like a warning that sounds like a threat. Being constructive is telling people to use a functional alternative to resolve their problem. Deleting those suggestions is significantly more spurious than me making them.

klattimer, 2015-08-07:
@johncstone try adding a 4k7 pullup resistor on the data pin, or if you already have a pull up there, remove it. Some of these parts work one way, some others depends on the manufacturer I guess. 

You'll have to be quick to read my comment though because trying to help people by suggesting wiringPi seems to lead to my comments being deleted. Then I get threatened for trying to be constructive and help people who have the same bug I hit.

klattimer, 2015-08-07:
@ladyada are you now SERIOUSLY editing MY words? That turns into misrepresentation. Absolutely disgusting!!! What a poisonous thing to do.

klattimer, 2015-08-07:
@johncstone please could you send me a copy of the original text I posted above as I am now seeking legal advice regarding being misrepresented by a third party without my consent. In the UK this will be covered under libellous activity.

klattimer, 2015-08-07:
@ladyada I suggest you contact me directly via email karl@qdh.org.uk to explain what the hell you think you're doing editing words which are attributed to me. This is illegal to do, you could use this power to misrepresent me in any number of ways and you should not have the power to do that. I have contacted github regarding this issue and am now in touch with a libel lawyer about getting this resolved via the courts.

trlaing, 2015-11-08:
I am having exactly the same issue. I installed Debian Jessie on my RPi2 before it was officially released. I'm pretty sure I got the image from here:
[link]https://images.collabora.co.uk/rpi2/[/link]
This image only uses one login (root) and does not support the "sudo" command. When logged in as root one has full control. I think this might be the issue, but then it might not have something installed to support the GPIO pins - I really am not sure. Here is the exact message I get:
[code]
root@WorkPi ~/Adafruit_Python_DHT/examples # python AdafruitDHT.py 11 4
Traceback (most recent call last):
  File "AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 90, in read_retry
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py", line 77, in read
  File "build/bdist.linux-armv7l/egg/Adafruit_DHT/Raspberry_Pi_2.py", line 34, in read
RuntimeError: Error accessing GPIO. Make sure program is run as root with sudo!
[/code]
It might be worth noting that the folder /root/Adafruit_Python_DHT/build/bdist.linux-armv7l does not include any files or folders. I installed AdafruitDHT using the instructions on the following web page:
[link]https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/software-install-updated[/link]
Here is the code I actually used:
[code]
cd ~
git clone https://github.com/adafruit/Adafruit_Python_DHT.git
cd Adafruit_Python_DHT
sudo apt-get update
sudo apt-get install build-essential python-dev python-openssl
sudo python setup.py install
cd examples
python AdafruitDHT.py 11 4
[/code]
Note that I did not use the command "sudo python AdafruitDHT.py 11 4" as sudo is not supported in my install:
[code]
root@WorkPi ~/Adafruit_Python_DHT/examples # sudo python AdafruitDHT.py 11 4
-bash: sudo: command not found
[/code]
I have just downloaded the official Debian Jessie image from raspberrypi.org, but do not wish to have to reinstall all my software again on the RPi2 as it is running sweet just now :-)
Any help would be much appreciated.

mvadu, 2015-12-03:
Refer to my implementation of pi_2_mmio.c @ https://github.com/AdysTech/raspy-weather/blob/master/dht11_driver/pi_2_mmio.c 

I had to change it to `/dev/gpiomem` from `/dev/mem` to get it to run without sudo as root.