adafruit/Adafruit_Python_DHT #21: Unable to open the sheet.

Shiva-Shinde, 2015-06-17:
Code runs perfectly fine with no errors etc. But it is unable to open the sheet. Please refer the code below. Unable to figure whats wrong here. Thanks!

import os
import ast
import requests, gspread
from oauth2client.client import SignedJwtAssertionCredentials
def authenticate_google_docs():
f = file(os.path.join('Shiva-c1c5516ecf00.p12'), 'rb')
SIGNED_KEY = f.read()
f.close()
scope = ['https://spreadsheets.google.com/feeds', 'https://docs.google.com/feeds']
credentials = SignedJwtAssertionCredentials('x@gmail.com', SIGNED_KEY, scope)
data = {
'refresh_token' : '241/ZLTD2DyOdiFJkN37mOZnmV-LtRu8wxY97o8Wn-nBkM4',
'client_id' : '2434760852908-5vi7v8ebilde26mvlran6ike982ovuhs.apps.googleusercontent.com', 
'client_secret' : '24EDI-HPmWBjvoTlm_NzJkcBmp',
'grant_type' : 'refresh_token',
}
r = requests.post('https://accounts.google.com/o/oauth2/token', data = data)
credentials.access_token = ast.literal_eval(r.text)['access_token']
gc = gspread.authorize(credentials)
return gc
gc = authenticate_google_docs()
import pdb; pdb.set_trace()
sh = gc.open("Hero")

I tried sh=gc.open("Hero").sheet1 (didn't work either)

tdicola, 2015-06-18:
Thanks for checking out the project!  Is there an error message or something being returned that might give more info on the problem?

ShivaShinde, 2015-07-22:
@tdicola Sorry for responding lately, I fixed this problems. I had some access_token issues. I am closing this issue. Thanks!!

closed 2015-07-22 by ShivaShinde