adafruit/Adafruit_Python_DHT #22: A FREQUENCY_SECONDS value greater than the API timeout results in an update every 2x FREQUENCY

kbresin, 2015-06-18:
Referring to examples/google_spreadsheet.py

I have a sample time set of 600 (ten minutes).

This results in an update every twenty minutes, as every second append requests always fails.

I suspect this is because the Google API login session expires in some time less than 5 minutes.

Two suggestions:

The sleep in the try/exception on line 125 should be static (e.g. 2), to allow a quicker retry.

At the very end of the script, I also added:

# force login if FREQUENCY_SECONDS is too high
if FREQUENCY_SECONDS > 120:
                worksheet = None

Which eliminates the errors in the first place.

I don't know the "ideal" value of 120.  I just know it is < 300.

Micchiu, 2015-07-06:
@kbresin 
Hello,
I have the exact same issue and I was thinking the same. Your comment confirms that could be the reason. I will try you solution and let you know.
'bye

tdicola, 2016-06-22:
Ah interesting issue, yeah ultimately oauth credentials can only live for a certain amount of time.  This is really more of a simple example to get started with writing data to google sheets.  I'll close this issue with the note that the frequency_seconds value should be kept below the oauth credential lifetime.

closed 2016-06-22 by tdicola