adafruit/Adafruit_Python_DHT #32: Incorrect data from AM2302

kropochev, 2015-10-31:
```
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Temp=12.0*  Humidity=3287.7%
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Temp=12.0*  Humidity=3287.7%
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Temp=12.0*  Humidity=3287.8%
```
Raspberry Pi B+
Raspbian Jessie 4.1.11
Real temp ~ 24*, hum ~22%

kropochev, 2015-10-31:
uncomment this line https://github.com/adafruit/Adafruit_Python_DHT/blob/master/source/Raspberry_Pi/pi_dht_read.c#L136
and compile

```
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Data: 0x0 0xe8 0x0 0xf4 0xdc
Temp=24.4*  Humidity=23.2%
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Data: 0x0 0xe8 0x0 0xf4 0xdc
Temp=24.4*  Humidity=23.2%
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Data: 0x0 0xe8 0x0 0xf0 0xdc
Data: 0x0 0xc7 0x0 0xf4 0xdb
Data: 0x0 0x60 0x1 0xe9 0xb8
Data: 0x0 0xe8 0x0 0xf4 0xdc
Temp=24.4*  Humidity=23.2%
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Data: 0x0 0xe8 0x0 0xf4 0xdc
Temp=24.4*  Humidity=23.2%
pi@raspberrypi1:~/Adafruit_Python_DHT/examples$ sudo ./AdafruitDHT.py 2302 18
Data: 0x0 0xe9 0x0 0xf3 0xdc
Temp=24.3*  Humidity=23.3%
```

benkrall, 2016-02-27:
one of my sensors is doing this - humidity reading is about 3289.4%
```
Data: 0x80 0x78 0x80 0x60 0x58
Data: 0x80 0x7e 0x0 0x61 0x60
Data: 0x80 0x7f 0x80 0x21 0x60
Data: 0x80 0x7f 0x80 0x61 0x60
Data: 0x80 0x7e 0x80 0x61 0x5f
Data: 0x80 0x7e 0x0 0x61 0x5f
Temp=9.7*  Humidity=3289.4%
```
the other sensor is located nearby, and working correctly
```
Data: 0x1 0x44 0x0 0x5f 0xa5
Data: 0x1 0x65 0x0 0x4e 0xb4
Temp=7.8*  Humidity=35.7%
```

Not sure what is going on - I have tested with 2 other sensors previously that worked perfectly...
Is the sensor busted??

MBradbury, 2016-05-02:
I am also having similar problems with a DHT22 on a Raspberry Pi 3. I've tested with another DHT22 and still see the same problem.

$ sudo ./AdafruitDHT.py 22 4
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Data: 0x81 0x38 0x0 0x4f 0x88
Failed to get reading. Try again!
$ sudo ./AdafruitDHT.py 22 4
Data: 0x81 0x3a 0x0 0x4e 0x89
Data: 0x81 0x3a 0x0 0x4e 0x89
Data: 0x81 0x3a 0x0 0x4e 0x89
Data: 0x81 0x3a 0x0 0x4e 0x89
Data: 0x81 0x3a 0x0 0x4e 0x89
Data: 0x81 0x3a 0x0 0x4f 0xa
Temp=7.9*  Humidity=3308.2%

A BMP085 is correctly reporting the temperature as 15.4*C

benkrall, 2016-05-02:
My problem was solved by adding the resistor as detailed in other schematics. I thought the resistor was built into the interior of this sensor and since it worked fine wired directly I did not add the resistor to the 3m cable.
Using a 3m cable required the resistor - once I installed it the sensor reports correctly.

MBradbury, 2016-05-02:
Sadly I was already using a 10K pull-up resistor, tried changing to 4.7K but it didn't improve anything. Cable length is about 10cm so shouldn't be an issue.

I get slightly better results with a DHT11 compared to the 16.9*C on my BMP085, but still not correct.
Temp=9.0*  Humidity=142.0%

tdicola, 2016-06-22:
Hrm it's tricky to troubleshoot this over github--unfortunately I can't repro a similar issue with my sensors.  There are a lot of variables like cable length, amount of other processes running at the same time (try not to run anything else, the code is a bit CPU-dependent), etc.  The best bet is to post in forums like http://forums.adafruit.com so more folks might see and be able to troubleshoot the setup.

closed 2016-06-22 by tdicola