adafruit/Adafruit_Python_DHT #33: Support for rpi board pin numbers

spuder, 2015-11-15:
Fron the examples and my testing, it appears that this library only supports using the GPIO.BCM pin numberings.

If you had an application that uses the GPIO.board numbering, it makes the code inconsistent. 

Is there a way to use this library with the raspberry pi board pin numbers?

AiGreek, 2016-02-01:
I agree with that. Can you do something ?
I tried but i couldn't fin where you set BCM

tdicola, 2016-06-22:
For now it only works with Pi BCM pin numbers--it's a little tricky since we can't use libraries like RPI.GPIO which support either number scheme.  Feel free to send a pull if anyone is interested in adding support though, thanks!

closed 2016-06-22 by tdicola