adafruit/Adafruit_Python_DHT #36: Support AM2301

SnouF, 2015-12-01:
It's possible to add support for AM2301 ?

mpainenz, 2016-03-01:
I have this same issue...

If you look at Adafruit_DHT\common.py, you will see the following:

```
# Define sensor type constants.
DHT11  = 11
DHT22  = 22
AM2302 = 22
SENSORS = [DHT11, DHT22, AM2302]
```

As you can see, the AM2302 uses the same code (22) as the DHT22. The AM2301 should use the same code as the DHT11.

Try setting your Sensor type to DHT11, it should work.

mpainenz, 2016-03-01:
According to the Data Sheets, the AM2301 and AM2302 use the same communication formula. 

[AM2301 Datasheet](https://kropochev.com/downloads/humidity/AM2301.pdf)
[AM2302 Datasheet](http://akizukidenshi.com/download/ds/aosong/AM2302.pdf)

So you can specify AM2302 for a AM2301.

closed 2018-11-02 by brennen