adafruit/Adafruit_Python_DHT #40: Can accessing gpio with sudo, still have Make sure program is run as root with sudo

amiceli, 2016-02-19:
I'ml trying to use my DHT11 sensor.
When I simpletest.py file, like this

`sudo python simpletest.py`

I always have this error : **Make sure program is run as root with sudo**

Thank you for your help

Mausy5043, 2016-02-20:
Try this:
`sudo simpletest.py`

Does that work?

amiceli, 2016-02-22:
No I've the same error.

tdicola, 2016-04-25:
Are you using the latest Raspbian Jessie release?  I can't repro the same issue with Jessie.  The code tries to do memory mapped access to the Pi's GPIO registers and if that fails then it can throw an error like you see warning to run as root with sudo.  If you're using a different Linux distro perhaps the root user doesn't have access to memory mapped GPIO (which would be really odd).  Try the latest Raspbian Jessie release, thanks!

closed 2016-04-25 by tdicola