adafruit/Adafruit_Python_DHT #41: Create compatibility with Jython?

ApprovedTOPKEK, 2016-03-20:
Hi!

I'm loading and running python scripts with Jython. Unfortunately, there seem to be compatibility issues between Jython and this library - I can't get scripts using this library to work with Jython.

Would there be a possiblity to resolve this problem?

Example of errors I get:
`Exception in thread "Thread-1" Traceback (most recent call last):
  File "Scripts/getTempScript.py", line 20, in retrieveData
    humidity, temperature = Adafruit_DHT.read_retry(Adafruit_DHT.AM2302, 4)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/common.py", line 90, in read_retry
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/common.py", line 76, in read
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/common.py", line 59, in get_platform
RuntimeError: Unknown platform.`

If I hardcode the Raspberry 2 Type in the get_platform function, I get this error:
`Exception in thread "Thread-1" Traceback (most recent call last):
  File "Scripts/getTempScript.py", line 20, in retrieveData
    humidity, temperature = Adafruit_DHT.read_retry(Adafruit_DHT.AM2302, 4)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/common.py", line 74, in read_retry
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/common.py", line 60, in read
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/common.py", line 42, in get_platform
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/Raspberry_Pi_2.py", line 22, in <module>
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/Raspberry_Pi_2_Driver.py", line 7, in <module>
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/Adafrut_DHT/Raspberry_Pi_2_Driver.py", line 6, in bootstrap
ImportError: No module named Adafruit_DHT.Raspberry_Pi_2_Driver`

Thing is, I don't know at all how to fix this. It is probably an error concerning Jython only, but it would be kind of awesome if there was a possibility to resolve it by changing the library, too.

Kind regards

ladyada, 2016-03-20:
nobody on our team uses jython - if you submit a pull request we'll take a look at merging it

ApprovedTOPKEK, 2016-03-26:
Unfortunately I don't think that I've enough knowledge to fix this. Just a thing I noticed though:

In the error message, the path sometimes says "Adafrut_DHT" instead of "Adafruit_DHT"; Could that be the cause of the problem?

For example here: 
File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.1.0-py2.7-linux-armv7l.egg/**Adafrut_DHT**/common.py"

Is this normal or did I make this happen?

tdicola, 2016-04-25:
What system are you running on, is it a Raspberry Pi or something else?  The code is actually platform-specific and has a C extension to very quickly read GPIO on the Pi.  If you try to run on other platforms beyond the Raspberry Pi & BeagleBone Black unfortunately it won't work because they don't have the optimizied code to read the sensor.  I have a feeling integrating with Jython will be difficult because the code uses Python C extensions and I'm not sure Jython supports it.  Your best bet would be to have Jython call the Python scripts here, for example the Adafruit_DHT.py in the examples could be called by a Jython script and its output parsed to grab the readings.  Again though it will only work on a Raspberry Pi (zero, 1, 2, 3) or BeagleBone Black.  Good luck!

closed 2016-04-25 by tdicola