adafruit/Adafruit_Python_DHT #44: Raspberry Pi 3 and Adafruit_Python_DHT

tcalbrecht, 2016-04-10:
Greetings.

I'm having a problem with the Adafruit DHT Python code.

- Installed all the required packages.
- sudo python setup.py install
- Connected the DHT11 signal to GPIO04 (pin 7).
- Tried to run:
    sudo ./AdafruitDHT.py 11 4
  and (after a few seconds) got:
    Failed to get reading. Try again!

The DHT11 sensor is working with C and the WiringPi library.

This is a Raspberry Pi 3, fresh Raspbian install, and all the latest software updates.
- pi@PI3:~/src/DHT11 $ uname -a
  Linux PI3 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

Thanks.

JockeNasvall, 2016-04-25:
I have the same problem. Any plans to support pi 3?

tdicola, 2016-04-25:
Hrm I can't repro any issues on the Pi 3, just tried reading an AM2302 and it works great with the current lib and the latest Raspbian Jessie.  Try installing a fresh SD card image and the library again to check there isn't something else going on.

Also make sure you don't have a lot of CPU intensive things running the background.  The library works by trying to very quickly read the output of the sensor but it can get tripped up if the kernel is interrupting it to let processor intensive tasks run.

And one last little thing to check is how you're powering the device.  Give it 3.3V power from the Pi instead of 5V power--the signal line has a pull-up resistor so you don't want to have it pull up to 5V and then feed that into an input on the Pi.

closed 2016-04-25 by tdicola

TiagoGouvea, 2016-12-21:
I'm getting on raspberry pi 3:

```Traceback (most recent call last):
  File "dht11_test.py", line 21, in <module>
    umid, temp = Adafruit_DHT.read_retry(sensor, pino_sensor);
  File "/home/pi/scripts/Adafruit_DHT/common.py", line 92, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/home/pi/scripts/Adafruit_DHT/common.py", line 78, in read
    platform = get_platform()
  File "/home/pi/scripts/Adafruit_DHT/common.py", line 53, in get_platform
    from . import Raspberry_Pi_2
  File "/home/pi/scripts/Adafruit_DHT/Raspberry_Pi_2.py", line 22, in <module>
    from . import Raspberry_Pi_2_Driver as driver
ImportError: cannot import name Raspberry_Pi_2_Driver````

even using --force-pi2 on setup

TiagoGouvea, 2016-12-21:
In fact it just my low knowlegde on python.. some related to folders and files locations... :(

yangl1996, 2017-01-05:
I'm working on RPi3 support here https://github.com/yangl1996/Adafruit_Python_DHT/tree/rpi3-support, stay tuned ;)

deepakkusahoo, 2017-07-01:
@TiagoGouvea  How did you manage to resolve this? I am also getting the exact same error.

manfye, 2017-09-03:
@TiagoGouvea i also have similar issue, how to solve it?

Cartmanishere, 2017-09-13:
For anyone else facing the same issue that @TiagoGouvea mentioned.
Try to run your program from outside the Adafruit_DHT directory that was cloned.
It happens because when you run -
`import Adafruit_DHT`
It imports the Adafruit_DHT folder in your current directory instead of the one installed in your site-packages directory. 
When run from anywhere else, it is working fine. 
Hope this helped.

teerna, 2018-01-18:
I got the same error in my R Pi 3 and I found a fix for it after some research.

So for those getting an error such as this one:
Traceback (most recent call last):
  File "simpletest.py", line 39, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 80, in read
    platform = get_platform()
  File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 55, in get_platform
    from . import Raspberry_Pi_2
  File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/Raspberry_Pi_2.py", line 22, in <module>
    from . import Raspberry_Pi_2_Driver as driver
ImportError: cannot import name Raspberry_Pi_2_Driver

Here is the fix(as per a comment by Mr Bogd):
1. Use `sudo python3 setup.py` install during the setup installation as you are more likely to be running Python 2 but the codes require Python 3.  

2. After the setup, run your .py file in any directory(home or any other) apart from the Adafruit_Python_DHT directory.
(Use the command `cp yourpythonfilename.py yourdirectoryname/` to copy the file to your desired directory where yourpythonfilename is the name of your python file and yourdirectoryname is the name of your directory you wish to copy the file to). 
This is because the python file imports the Adafruit_DHT directory(through this line: `import Adafruit_DHT`) so it should not be in the same directory.

Lynexo, 2018-03-16:
How exactly do I do this @Cartmanishere ?

LucasAmos, 2018-04-26:
I had this issue running a Raspberry Pi using resin.io
I was running `python setup.py install --force-pi` instead of `python setup.py install --force-pi2` even though I am using a Raspberry Pi 3.

Very poor documentation for this adafruit library

geofbaum, 2018-04-26:
@LucasAmos What issue are you alluding to? I have yet to see any issues loading and running this library on an RPi2 or RPi3? You mention using resin.io which has nothing to do with this library and then say it's poorly documented?

edgarbc, 2018-09-12:
Thank you! This worked for me.
 
> I got the same error in my R Pi 3 and I found a fix for it after some research.
> 
> So for those getting an error such as this one:
> Traceback (most recent call last):
> File "simpletest.py", line 39, in
> humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
> File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 94, in read_retry
> humidity, temperature = read(sensor, pin, platform)
> File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 80, in read
> platform = get_platform()
> File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 55, in get_platform
> from . import Raspberry_Pi_2
> File "/home/pi/Adafruit_Python_DHT/Adafruit_DHT/Raspberry_Pi_2.py", line 22, in
> from . import Raspberry_Pi_2_Driver as driver
> ImportError: cannot import name Raspberry_Pi_2_Driver
> 
> Here is the fix(as per a comment by Mr Bogd):
> 
>     1. Use `sudo python3 setup.py` install during the setup installation as you are more likely to be running Python 2 but the codes require Python 3.
> 
>     2. After the setup, run your .py file in any directory(home or any other) apart from the Adafruit_Python_DHT directory.
>        (Use the command `cp yourpythonfilename.py yourdirectoryname/` to copy the file to your desired directory where yourpythonfilename is the name of your python file and yourdirectoryname is the name of your directory you wish to copy the file to).
>        This is because the python file imports the Adafruit_DHT directory(through this line: `import Adafruit_DHT`) so it should not be in the same directory.

Gaurav70042, 2018-09-18:
this error is annonying ..... I have to my project tommorow morning!!!!!