adafruit/Adafruit_Python_DHT #52: Raspberry 1 pi model B uncompatible with AM2302

yagarea, 2016-07-27:
Raspberry 1 pi model B uncompatible with AM2302

Traceback (most recent call last):
  File "./AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 90, in read_retry
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 77, in read
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/Raspberry_Pi.py", line 34, in read
RuntimeError: Error accessing GPIO.

It works with old commit  9e8109bb4ab5ec9127e53e792c1f69eddfd2f687 .

ethanhjennings, 2016-08-01:
Same problem and exception here with my DHT22 and Raspberry Pi 1 model B. 

It also works with the same commit (9e8109bb4ab5ec9127e53e792c1f69eddfd2f687) mentioned by @yagarea.

icuy, 2016-12-31:
Same problem and exception here with my AM2302 and Raspberry Pi 1 model B+.

aequitas, 2017-01-29:
Same problem with DHT11 and Raspberry Pi 1 model B. But does also work on older version: https://github.com/adafruit/Adafruit_Python_DHT/commit/9e8109bb4ab5ec9127e53e792c1f69eddfd2f687

aequitas, 2017-01-29:
I suspect this change broke it: https://github.com/adafruit/Adafruit_Python_DHT/commit/9297cd60f07fb2dfcad7ac7b9b525c7ec1401e31

Looking in my `/dev` there is no `gpiomem` only `mem`.

aequitas, 2017-01-29:
I have created a solution which fall's back to the old behavior of using `/dev/mem` if `/dev/gpiomem` is unavailable. https://github.com/adafruit/Adafruit_Python_DHT/pull/56

brennen, 2018-07-27:
#56 merged, so I think this is good at present.

closed 2018-07-27 by brennen