adafruit/Adafruit_Python_DHT #53: DHT11 RPI Negative Values not parsed / displayed

etlmwwn, 2016-11-07:
Using the standard libraries for DHT 11 / RPI / Python and GDocs (https://learn.adafruit.com/dht-humidity-sensing-on-raspberry-pi-with-gdocs-logging/overview) works perfectly.

The issue is that when the DHT sensor encounters temperatues below 0 degrees, the negative values are not captured or displayed. This same issue is encountered when the simpletest.py, AdafruitDHT.py or google_spreadsheet.py (yes - tested DHT11 in the freezer....), the lowest value captured was 1.

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  

Raspberry Pi with Raspbian
Raspberry Pi reference 2016-09-23
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 62406bad92ed23728f46711b3539c04c37dfb62c, stage2

Linux rpi1 4.4.21+ #911 Thu Sep 15 14:17:52 BST 2016 armv6l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"

- Python version (run `python -version` or `python3 -version`): 

Python 2.7.9

- Error message you are receiving, including any Python exception traces: 

No errors or traces, output reported as above

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

Put DHT11 in Freezer for 10 minutes, and keep running simple.py or the other standard scripts.

pi@rpi1:~/Adafruit_Python_DHT/examples $ sudo ./simple.py
Temp=1.0*C  Humidity=18.0%

Lowest value of 1.0*C reported.

>> Similar issue reported here way back when: https://www.raspberrypi.org/forums/viewtopic.php?t=62467&p=464881

MeraX, 2016-11-08:
The DHT11 is specified for 0 to 50°C

etlmwwn, 2016-11-08:
Cheers for clarifying. Need to pay more attention to the HW Specifications :)

closed 2016-11-08 by etlmwwn