adafruit/Adafruit_Python_DHT #54: Using the built-in pull-up resistor in Raspberry Pi2.

kukukk, 2017-01-13:
Hi,

I want to use a DHT22 sensor connected to the GPIO port of a RPi2. Is it possible to activate the built-in pull-up resistor?
The following code will set the pull-up resistor or Adafruit_DHT.read_retry will reset it?
```
import Adafruit_DHT
import RPi.GPIO as GPIO

sensor_pin = 4
sensor_type = Adafruit_DHT.DHT22

GPIO.setmode(GPIO.BCM)
GPIO.setup(sensor_pin, GPIO.IN, GPIO.PUD_UP)

humidity, temperature = Adafruit_DHT.read_retry(sensor_type, sensor_pin)
```
If the pull-up resistor is not working, humidity and temperature will be None or they will just have an incorrect value, like 23% and 24°C instead of 64% and 19°C?

Thanks

piecyk, 2017-11-22:
@kukukk any news about it ?

kukukk, 2017-11-28:
Nope, but I'm no longer using this library to read the values.

BernardoSilva, 2017-11-28:
@kukukk  what library are you using now, just for curiosity? :)

kukukk, 2017-11-28:
Hi @BernardoSilva,

I'm using OSMC on my RPI2 and it supports device-tree. So, I just enable the device-tree for DHT11 in /boot/config.txt by adding
`dtoverlay=dht11,gpiopin=4`
Where gpiopin is the pin number where my sensor is connected. It works with the DHT22 sensor too.

This way all I have to do is to read the content of the following 2 files:
`/sys/devices/platform/dht11@0/iio:device0/in_temp_input`
`/sys/devices/platform/dht11@0/iio:device0/in_humidityrelative_input`

BernardoSilva, 2017-11-28:
Thank you for sharing @kukukk =D

Will have a look to try with my DHT11.

timdonovanuk, 2017-12-22:
It's not clear if using the device tree method negates the need for a pull-up resistor. Anyone know?

kukukk, 2017-12-22:
If I remember correctly the device tree method does not activate the pull-up resistor.

I don't have an external pull-up resitor and I don't know whether I have the original dht11 overlay file or a modified one, but my readings seems correct.
(I tried to activate the built-in pull-up resistor by downloading the dht11-overlay.dts, modifying it in order to activate the pull-up resistor and rebuild it to get the new dht11-overlay.dtb.)

I don't know whether the missing pull-up resistor would cause reading errors or just incorrect values.

timdonovanuk, 2017-12-22:
Thank you @kukukk ! At the moment I just get a "read error" when I cat the iio:device0/in_temp_input file but I'll play around and see if I can get it working.

kukukk, 2017-12-22:
What I can suggest are the standard advices:
  - check wether you properly connected the sensor
  - check if you have proper voltage on the sensor's pins, because with long cables it can be a problem
  - check whether you configured the right GPIO port in the config.txt file

You can also check whether you get a proper reading with this Adafruit Python DHT library first before trying the device tree solution.

timdonovanuk, 2017-12-22:
Just getting my soldering iron out to make some more permanent connections! Thanks :)

dade80vr, 2018-07-24:
Hi guys, i'm also interested in pull up resistor using this library.
Have you got any update? 
Is it possible to activate via `GPIO.PUD_UP` or it will reset by the library?
Thank you.

ladyada, 2019-11-08:
hi this repo is going to be deprecated soon, please try 
https://github.com/adafruit/Adafruit_CircuitPython_DHT
which will replace this library

closed 2019-11-08 by ladyada