adafruit/Adafruit_Python_DHT #58: Usage of DHT reading function interferes/disturbs with the running script

kevins2016, 2017-02-10:
Hi,

I have an issue with the usage of the read_retry(sensor, pin) function.

I have imported Adafruit_DHT into a python script and I am using the read function from this script with:
`humidity, temperature = Adafruit_DHT.read_retry(Adafruit_DHT.DHT22, gpiopin)`

It reads out values correctly but whenever the function is called the whole program slows down heavily until the function is finished.
To clarify this, the program is running several threads and all running threads and the main slow down.

I just wrote a little script where you can see the issue:
```
import Adafruit_DHT # adafruit dht module for temp/humi sensor reading
import time
import threading

class Thread(threading.Thread): #class to enable threading
        def __init__(self, target, *args):
                self._target = target
                self._args = args
                threading.Thread.__init__(self)
        def run(self):
                while 1:
                        self._target(*self._args)

def read3dhts(): #read dhts every 2 seconds
        time.sleep(2)
        print ("reading")
        humidity1, temperature1  = Adafruit_DHT.read_retry(Adafruit_DHT.DHT22, 16)
        humidity2, temperature2  = Adafruit_DHT.read_retry(Adafruit_DHT.DHT22, 20)
        humidity3, temperature3  = Adafruit_DHT.read_retry(Adafruit_DHT.DHT22, 21)

t1= Thread(read3dhts)
t1.daemon = True
t1.start() #run dht reading in a thread

while 1:
        time.sleep(.1)
        print ("Program Running")
```

If you run it you can see that the printing of "Program Running" significantly slows down when reading the sensor even though the reading runs in an own thread. I have read three of them to make the effect bigger, but if you only read one it is basicly the same. If you read out a pin that doesnt have a sensor at all it also slows down a little bit.
However when I take out the sensor reading function from my program and call the function from a second python script it does not slow down the original program anymore.

Any ideas how to solve this?
thanks in advance

kevins2016, 2017-02-13:
I have found the reason. In the pi_2_dht_read.c source file theres a line
`pi_2_mmio_set_high(pin);
sleep_milliseconds(500);`

the sleep_miliseconds command causes the delay. I changed it with the standard C command sleep(); and now it runs fine.

closed 2017-02-13 by kevins2016

pioto, 2018-05-04:
@kevins2016 any chance you want to propose a pull request with your fix? I'm seeing a similar problem with a similar sort of "worker thread w/ sleeps between each poll" setup.