adafruit/Adafruit_Python_DHT #61: 4.9.11+ - Unkown platform

retikulumx, 2017-02-22:
Issues with the current Raspbian update:
Linux raspberrypi 4.9.11+

`Traceback (most recent call last):
  File "./AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 90, in read_retry
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 76, in read
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 59, in get_platform
RuntimeError: Unknown platform.
`

Thanks for fixing it.

retikulumx, 2017-02-22:
Seems like it is a bug in the new raspbian. uname -p returns "unknown"

retikulumx, 2017-02-22:
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0010
Serial          : 00000000e6858719

retikulumx, 2017-02-22:
Ok, solved. My brother and me (more my brother than me) found the problem:

/proc/cpuinfo gives BCM2835 now since the kernel update to 4.9 . Raspbian changed some naming here.
So what you have to do is, open the platform_detect.py, scroll down to the bottom and the find line

`   if match.group(1) == 'BCM2708':
        # Pi 1
`

and change it to

`   if match.group(1) == 'BCM2835':
        # Pi 1
`
(or the Pi 2 line if the isse happens here too)
Don't forget to reinstall! That's it.

ProhtMeyhet, 2017-02-22:
some explanation on the different naming maybe found here

https://raspberrypi.stackexchange.com/questions/840/why-is-the-cpu-sometimes-referred-to-as-bcm2708-sometimes-bcm2835

ProhtMeyhet, 2017-02-22:
and here is the remove commit in linux-4.8:

> Drop ARCH_BCM2708 and ARCH_BCM2709 
> They are not necessary anymore since both are based on ARCH_BCM2835.
> Also use the compatible strings "brcm,bcm2835", "brcm,bcm2836" and "brcm,bcm2837".

https://github.com/raspberrypi/linux/commit/3ad156116a1e6e3e9ad5a9a22887465b75b3b62d

retikulumx, 2017-02-22:
> raspberrypi/linux@3ad1561

Aaaah ok, cool. Danke!

tdicola, 2017-02-28:
Ah interesting the cpu info name changed with the later kernel.  I just merged in a pull that should handle this though, so try grabbing the library again and reinstalling (might need to run `sudo python setup.py clean --all` then `sudo python setup.py install --force` to force a recompile and full reinstall).  Thanks for raising this!

closed 2017-02-28 by tdicola

retikulumx, 2017-03-01:
Fix validated with Raspberry Zero.

Thanks!

TheCrimsonIdol, 2017-04-13:
I solved it by fetching latest code from repository here, and running just install by: `sudo python setup.py install`
It worked like a charm afterwards. Thanks!

thomasbreitbach, 2017-05-06:
Thanks for the fix! It works. 👍

franzda, 2017-05-06:
Maybe I have not understood something yet. I´m using RPi 3B. With Phython 2.7.9 and Python 3.4.2. 
What do I have to install now to address DHT22 via Adafruit.

stamster, 2017-05-18:
@franzda you need to pull latest version of adafruit code from this repo, and to run setup again (install).

GEARWELL, 2019-09-17:
To solve the problem in Ubuntu mate, i followed the instructions by retikulumx .

I edited the file  platform_detect.py  and change the lines
    with open('/proc/cpuinfo', 'r') as infile:
with my own cpuinfo file
    with open('/your_location/cpuinfo', 'r') as infile:

Thaths because i only have in my cpuinfo:

processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

And it must be like:

  GNU nano 2.9.3                                  cpuinfo

processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware : BCM2835
Revision : 0010
Serial : 00000000e6858719

Just added the

Hardware : BCM2835
Revision : 0010
Serial : 00000000e6858719

and saved in other location to solve the Unknown Platform errror