adafruit/Adafruit_Python_DHT #63: Raspberry Pi GPIO17 with DHT22 no longer working

kowenhouston, 2017-03-09:
Hi There, 

I have a few raspberry pi's wired with DHT22 sensors. It would be very expensive for me to wire them all back up. 

This is a working raspberry pi with raspbian:
- 4.1.20-v7+ #867 SMP Wed Mar 23 20:12:32 GMT 2016 armv7l GNU/Linux
- Debian 7.11
- Python 2.7.3
```
root@localhost:~# python AdafruitDHT.py 22 17
Temp=24.2*  Humidity=29.9%
```

The non working raspberry pi with rasbian is:
- 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux
- Debian 7.11
- Python 2.7.3
```
root@localhost:~# python AdafruitDHT.py 22 17
Traceback (most recent call last):
  File "AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 90, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 77, in read
    return platform.read(sensor, pin)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/Beaglebone_Black.py", line 206, in read
    raise ValueError('Pin must be a valid GPIO identifier like P9_12 or GPIO1_28.')
ValueError: Pin must be a valid GPIO identifier like P9_12 or GPIO1_28.
```

The only difference between the two Raspberry Pis is that I ran **rpi-update** on one, which is now not working. The are both wired up identically, have the identical image (however I updated one).

My Wiring looks like this:
GPIO17 / Raspberry Pi Pin 11 - DHT22 Pin 2
+5VDC / Raspberry Pi Pin 2 - DHT22 Pin 1
GND / Raspberry Pi Pin 6 - DHT22 Pin 4

ladyada, 2017-03-09:
hiya, yep some serious breaking changes in latest raspbian. best suggestion is to go back to the previous version for now!

kowenhouston, 2017-03-09:
Hey, The reason for updating was because I wanted to be able to use the Raspberry Pi 2 v1.2, Is there a version that works both on the Raspberry Pi 2 v1.2 and will allow me to use this library on GPIO 17?

I tried dropping back to this version
> Commits on Sep 22, 2016
>  @popcornmix
> platform: Enable support for Pi2+
> 21505f3fb5d8e2a5abbf95baf441a7a6415d4a82

```
rpi-update 21505f3fb5d8e2a5abbf95baf441a7a6415d4a82
```

Which still gives me an error, but it's different again.
```
python AdafruitDHT.py 22 17
Traceback (most recent call last):
  File "AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 90, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 76, in read
    platform = get_platform()
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 51, in get_platform
    import Raspberry_Pi_2
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/Raspberry_Pi_2.py", line 22, in <module>
    import Raspberry_Pi_2_Driver as driver
ImportError: No module named Raspberry_Pi_2_Driver
```

kowenhouston, 2017-03-09:
Ok, so I figured out the answer. This should allow the GPIO port 17 to work with the DHT sensor and the image to be used on the Raspberry Pi 2 v1.2

Downgrade the version to 4.4.50-v7+:
> Commits on Feb 21, 2017
>  @popcornmix
> kernel: Bump to 4.4.50

```
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764
sudo reboot
```

Then when the Pi is back online make sure the build packages are up to date:
```
sudo apt-get install build-essential python-dev
```
Uninstall the DHT Package:
```
sudo pip uninstall Adafruit_Python_DHT
```
Remove all cached DHT Packages
```
sudo rm -Rf /usr/local/lib/python2.7/dist-packages/Adafruit_*
```
And reinstall the DHT Package:
```
sudo pip install Adafruit_Python_DHT
```

Result:
```
root@localhost:~# python AdafruitDHT.py 22 17
Temp=28.1*  Humidity=46.2%
```

- 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
- Debian 7.11
- Python 2.7.3

closed 2017-03-09 by kowenhouston

manfye, 2017-09-03:
hi im using pi 3, I have the similar issue ImportError: No module named Raspberry_Pi_2_Driver, how to solve it?

Markkop, 2017-09-05:
Hey! I've intensively searched for information and such for these errors and I've come to a conclusion that this library doesn't work correctly in Raspberry Pi 3, right?
I've bough one last week and couldn't make it work with DHT11, however it worked fine with other sensors (and codes).
It seems the solution is downgrading its version, but I'm afraid this could affect RPi somehow.
Could I ask for an update? :p

joshuaboniface, 2017-10-12:
I found this issue was happening to every GPIO on the latest RPi3 version - got the RPi3 a month ago, and today tried to hook up 4 AM2302 sensors to every GPIO I could - none of them would work. But after following @kowenhouston 's advice, they're all working perfectly. Hopefully these keywords help the next person find this issue on Google!

Twisis, 2018-03-20:
I have the same issue I believe. 

The DHT22.py file reports this error:  ...Non-ASCII character '\xa3'... (related to line 86)

line 86 of DHT22.py:  temperature = int_t          #dht11 0..50*C

How can I resolve this issue? I had a few other errors which were resolved with proper indentation
of the python code (I didn't know python was this strict with it's syntax). I followed a youtube tutorial to try and get the sensor working. I got the DHT22.py code from github, placed it in the same folder as the sensor program and used it as an import in the sensor program. But what I got was errors unlike in the tutorial.

Twisis, 2018-03-20:
...I took down all the comments.
Then it reported an error on the first line, that it couldn't find pyb from the import pyb line.
And then I found it !

...I get different code when I open the file from my windows computer than I do when open it from the raspi ?!? It's the same zip file from the same url address ?!?

geofbaum, 2018-03-21:
@Twisis what DHT22.py file are you using? There is no DHT22.py file in this repository so I don't know which repo you got your file from. If you can give me the link to the file's repository I can take a look at the original code. But without seeing that, there's not much to work with since as I said yesterday the error refers to the # symbol but your response then goes on to talk about an error referring pyb or a micropython file that is again not within this repository.

If you can point me in the direction of the files you are using here on github I'd be happy to take a look at them and see if I can be of more help.

Twisis, 2018-03-22:
This is the address of the file used: http://abyz.me.uk/rpi/pigpio/code/DHT22_py.zip

When I used the link through the Raspberry Pi, opened the zip file and then opened the DHT22.py text file I got the following code:

`import pyb
from pyb import Pin
from pyb import ExtInt

# We need to use global properties here as any allocation of a memory (aka declaration of a variable)
# during the read cycle causes non-acceptable delay and we are loosing data than
data = None
timer = None
micros = None
dhttype = 0  #0=DHT11 1=DHT21/DHT22

FALL_EDGES = 42 # we have 42 falling edges during data receive

times = list(range(FALL_EDGES))
index = 0

# The interrupt handler
def edge(line):
    global index
    global times
    global micros
    times[index] = micros.counter()
    if index < (FALL_EDGES - 1): # Avoid overflow of the buffer in case of any noise on the line
        index += 1

def init(timer_id = 2, data_pin = 'Y2', the_dhttype='DHT22'):
    global data
    global micros
    global timer
    global dhttype 
	 
    if(the_dhttype=='DHT11'):
	dhttype = 0
    else:
	dhttype = 1
    
    # Configure the pid for data communication
    data = Pin(data_pin)
    # Save the ID of the timer we are going to use
    timer = timer_id
    # setup the 1uS timer
    micros = pyb.Timer(timer, prescaler=83, period=0x3fffffff) # 1MHz ~ 1uS
    # Prepare interrupt handler
    ExtInt(data, ExtInt.IRQ_FALLING, Pin.PULL_UP, None)
    ExtInt(data, ExtInt.IRQ_FALLING, Pin.PULL_UP, edge)

# Start signal
def do_measurement():
    global data
    global micros
    global index
    # Send the START signal
    data.init(Pin.OUT_PP)
    data.low()
    micros.counter(0)
    while micros.counter() < 25000:
        pass
    data.high()
    micros.counter(0)
    while micros.counter() < 20:
        pass
    # Activate reading on the data pin
    index = 0
    data.init(Pin.IN, Pin.PULL_UP)
    # Till 5mS the measurement must be over
    pyb.delay(5)
[... 309 lines cut ...]

   next_reading = time.time()

   while True:

      r += 1

      s.trigger()

      time.sleep(0.2)

      print("{} {} {} {:3.2f} {} {} {} {}".format(
         r, s.humidity(), s.temperature(), s.staleness(),
         s.bad_checksum(), s.short_message(), s.missing_message(),
         s.sensor_resets()))

      next_reading += INTERVAL

      time.sleep(next_reading-time.time()) # Overall INTERVAL second polling.

   s.cancel()

   pi.stop()

`

Twisis, 2018-03-22:
import pyb
from pyb import Pin
from pyb import ExtInt

# We need to use global properties here as any allocation of a memory (aka declaration of a variable)
# during the read cycle causes non-acceptable delay and we are loosing data than
data = None
timer = None
micros = None
dhttype = 0  #0=DHT11 1=DHT21/DHT22

FALL_EDGES = 42 # we have 42 falling edges during data receive

times = list(range(FALL_EDGES))
index = 0

# The interrupt handler
def edge(line):
    global index
    global times
    global micros
    times[index] = micros.counter()
    if index < (FALL_EDGES - 1): # Avoid overflow of the buffer in case of any noise on the line
        index += 1

def init(timer_id = 2, data_pin = 'Y2', the_dhttype='DHT22'):
    global data
    global micros
    global timer
    global dhttype 
	 
    if(the_dhttype=='DHT11'):
	dhttype = 0
    else:
	dhttype = 1
    
    # Configure the pid for data communication
    data = Pin(data_pin)
    # Save the ID of the timer we are going to use
    timer = timer_id
    # setup the 1uS timer
    micros = pyb.Timer(timer, prescaler=83, period=0x3fffffff) # 1MHz ~ 1uS
    # Prepare interrupt handler
    ExtInt(data, ExtInt.IRQ_FALLING, Pin.PULL_UP, None)
    ExtInt(data, ExtInt.IRQ_FALLING, Pin.PULL_UP, edge)

# Start signal
def do_measurement():
    global data
    global micros
    global index
    # Send the START signal
    data.init(Pin.OUT_PP)
    data.low()
    micros.counter(0)
    while micros.counter() < 25000:
        pass
    data.high()
    micros.counter(0)
    while micros.counter() < 20:
        pass
    # Activate reading on the data pin
    index = 0
    data.init(Pin.IN, Pin.PULL_UP)
    # Till 5mS the measurement must be over
    pyb.delay(5)

# Parse the data read from the sensor
def process_data():
    global dhttype
    global times
    i = 2 # We ignore the first two falling edges as it is a respomse on the start signal
    result_i = 0
    result = list([0, 0, 0, 0, 0])
    while i < FALL_EDGES:
        result[result_i] <<= 1
        if times[i] - times[i - 1] > 100:
            result[result_i] += 1
        if (i % 8) == 1:
            result_i += 1
        i += 1
    [int_rh, dec_rh, int_t, dec_t, csum] = result

    if dhttype==0:		# dht11		
	humidity = int_rh 	
	# dht11 20% ~ 90%

	temperature = int_t 	
	# dht11 0..50°C

	# dht21,dht22
	else:	
	humidity = ((int_rh * 256) + dec_rh)/10 
	temperature = (((int_t & 0x7F) * 256) + dec_t)/10
	if (int_t & 0x80) > 0:
		temperature *= -1
         
    comp_sum = int_rh + dec_rh + int_t + dec_t
    if (comp_sum & 0xFF) != csum:
        raise ValueError('Checksum does not match')
    return (humidity, temperature)

def measure():
    do_measurement()
    if index != (FALL_EDGES -1):
        raise ValueError('Data transfer failed: %s falling edges only' % str(index))
    return process_data()

#using:
#import DHT22
#DHT22.init()
#DHT22.measure()

Twisis, 2018-03-22:
I don't know why the code looks like that when I add it.

Twisis, 2018-03-22:
These are the two different DHT22.py files.

geofbaum, 2018-03-22:
First thing of note, that file is not from github, as you said in your initial post.

Secondly, if you are posting python code you should use the Markdown styling:
````
    ```python
    <your code here>
    ```
````
With your code in between it.

Thirdly the first of the two files you copied and pasted the code above are not from that link you shared. Even if you think it's the same file and it is what you see on your Raspberry Pi there's no way it is from the same link. There are references of the dht11 in the first set of code while there is nothing in the second version that you say is the same file opened in Windows. That first set of code once again refers to the Micropython project library `https://github.com/micropython` and if you are trying to run code from that you should try running it with the Adafruit MicroPython tool:

`https://learn.adafruit.com/micropython-basics-load-files-and-run-code?view=all`

I'm pretty sure this mix up and blending of code is where you're problem is stemming from. You are somehow mixing multiple files that are not alike and expecting them to run through a regular python build. This would make sense since you received an error about pyb, especially since you probably don't have that library installed or if you do it's not within your regular python lib directory.

jevescio, 2018-03-30:
I am at Wits end.. HELP.   I have tried everything to get a DHT22/AM2302 working on a Pi 3B+ with latest Stretch loaded.  I have not downgraded as mentioned above yet and that will be next if someone says it will work on latest Pi.  I followed Tony D video and code from https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code.git.
Even formatted card with fresh OS and reinstalled required code.

2.7 and 3.5 return same results  " Failed to get reading. Try Again!"

geofbaum, 2018-03-30:
@jevescio So you are using the code from this repository since the link you shared is to the outdated Adafruit Repo that links to this one?

Also you aren't getting an error like what folks above were getting. The library is working correctly as it was written, you are just not catching a reading which is why it's giving you that result.  Can you post the exact code that you are using?

jevescio, 2018-03-30:
@geofbaum I am not getting same error seems to be beyond the above issues but still not reading sensor.  I am just using the example code now to try simple read to see if working.
https://github.com/adafruit/Adafruit_Python_DHT
installed from there also setup.py for both 2.7 and 3.5

i even tried the  http://abyz.me.uk/rpi/pigpio/examples.html#Python%20code before i formatted and started over.

geofbaum, 2018-04-08:
@Twisis  and @jevescio I can only assume that you guys have fixed the problems you had or have given up regarding the issues you faced?

jevescio, 2018-04-09:
@geofbaum Been out of town and will test your suggestion tonight.  I also got new DHT22 just to try in-case the other is bad.   Will let you know findings.
Thanks for all your help

geofbaum, 2018-04-09:
@jevescio No worries, just wanted to see how things had progressed if they had. Hopefully when you test it, things will work out better!

Twisis, 2018-04-09:
I got it working. The issue was I received two different texts of code when using the raspberry pi vs. my pc.
Why? I have no clue.

geofbaum, 2018-04-09:
@Twisis Good to hear you got it working.

jevescio, 2018-04-09:
@geofbaum I got this working with the new sensors.  I must have had 2 bad sensors

Again, thanks for all your support

geofbaum, 2018-04-09:
@jevescio Good to hear and you're welcome! Glad to be able to help when I can.