adafruit/Adafruit_Python_DHT #68: No attribute 'read_retry'

TTN-, 2017-05-10:
**Problem:**
```
Traceback (most recent call last):
  File "test_read.py", line 4, in <module>
    print(Adafruit_Python_DHT.read_retry(Adafruit_Python_DHT.DHT22, 27))
AttributeError: 'module' object has no attribute 'read_retry'
```

- Python version Python 3.4.2

**Steps to reproduce:**
```
git clone https://github.com/adafruit/Adafruit_Python_DHT.git
cd  Adafruit_Python_DHT
```
I require it to work on python3, so we will install to python 2 and 3:
```
sudo python setup.py install
sudo python3 setup.py install # <---- note I'm installing to python 2  and 3
```
Verifying we can read the sensor:
```
cd ./Adafruit_Python_DHT/examples
sudo ./AdafruitDHT.py 22 27
Temp=20.0*  Humidity=63.1%
```
Ok, so far so good.
Now, lets run it with python 3:
```sysop@kivypie:~/Adafruit_Python_DHT/examples$ python3 AdafruitDHT.py 22 27
Temp=19.8*  Humidity=63.6%
```
Now lets verify we're actually using python 3 inside the file we're running:
```
nano ./AdafruitDHT.py
```
Add:
```
import sys
print (sys.version)
```
Now lets run it again to make sure we're actually using python 3:
```
sysop@kivypie:~/Adafruit_Python_DHT/examples$ python3 AdafruitDHT.py 22 27
3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1]
Temp=19.8*  Humidity=63.7%
```
Perfect. So reading the sensor works on python3.

Now lets try it again, but now in a test file:
```
import Adafruit_Python_DHT
print(Adafruit_Python_DHT.read_retry(Adafruit_Python_DHT.DHT22, 27))
```
thats in `test_read.py` in the home folder, and the Adafruit_Python_DHT folder is also in the home folder.
```
sysop@kivypie:~$ python3 test_read.py 
Traceback (most recent call last):
  File "test_read.py", line 4, in <module>
    print(Adafruit_Python_DHT.read_retry(Adafruit_Python_DHT.DHT22, 27))
AttributeError: 'module' object has no attribute 'read_retry'```

So the question is, why does it work on python3 in the example folder but not in my own file?

TTN-, 2017-05-10:
SOLVED: I had to use:
```
import Adafruit_DHT
print(Adafruit_DHT.read_retry(Adafruit_DHT.DHT22, 27))
```
So for people wanting to use it on python 3, instructions are above, and use as in the snippet ^

closed 2017-05-10 by TTN-

Walter1White, 2017-07-15:
Python 3.6.1 (v3.6.1:69c0db5, Mar 21 2017, 17:54:52) [MSC v.1900 32 bit (Intel)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
 RESTART: C:\Users\Walter\AppData\Local\Programs\Python\Python36-32\MUS CONTROL with adafruit.py 
Traceback (most recent call last):
  File "C:\Users\Walter\AppData\Local\Programs\Python\Python36-32\MUS CONTROL with adafruit.py", line 21, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(Adafruit_DHT.AM2302, 4)
AttributeError: module 'Adafruit_DHT' has no attribute 'read_retry'
>>>