adafruit/Adafruit_Python_DHT #71: Non-functional on Raspberry Pi B+ on Raspbian stretch

pioto, 2017-08-31:
It seems that this library may not work on Raspbian stretch; I recently upgraded from jessie, and it seems it hasn't worked since then with my AM2302.

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry Pi B+ (Rev 0010), Raspbian 9.1 (stretch)

- Python version (run `python -version` or `python3 -version`):  Python 2.7.13

- Error message you are receiving, including any Python exception traces:

```
pi@rpi-01:~/git/Adafruit_Python_DHT$ time python ./examples/simpletest.py                                                                   
Failed to get reading. Try again!

real    0m44.628s
user    0m6.430s
sys     0m0.410s
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): Simply try running the included `simpletest.py`, adjusted for my particular hardware setup:

```diff
pi@rpi-01:~/git/Adafruit_Python_DHT$ git diff
diff --git a/examples/simpletest.py b/examples/simpletest.py
index 06c4d9e..9849772 100644
--- a/examples/simpletest.py
+++ b/examples/simpletest.py
@@ -24,15 +24,17 @@ import Adafruit_DHT
 
 # Sensor should be set to Adafruit_DHT.DHT11,
 # Adafruit_DHT.DHT22, or Adafruit_DHT.AM2302.
-sensor = Adafruit_DHT.DHT22
+#sensor = Adafruit_DHT.DHT22
+sensor = Adafruit_DHT.AM2302
 
 # Example using a Beaglebone Black with DHT sensor
 # connected to pin P8_11.
-pin = 'P8_11'
+#pin = 'P8_11'
 
 # Example using a Raspberry Pi with DHT sensor
 # connected to GPIO23.
 #pin = 23
+pin = 4
 
 # Try to grab a sensor reading.  Use the read_retry method which will retry up
 # to 15 times to get a sensor reading (waiting 2 seconds between each retry).
```

ladyada, 2017-08-31:
@PaintYourDragon heya here's one of those GPIO C changeovers!

pioto, 2017-08-31:
Let me know if there's any extra tests or diagnostics you need from me to help fix this.

pioto, 2017-09-02:
I'm sorry, looks like a small "helper" was playing with my breadboard around when I had done my Raspbian upgrade... Once I corrected the wiring, everything was good.

closed 2017-09-02 by pioto