adafruit/Adafruit_Python_DHT #73: Can't Read Two Sensor

harlilopez, 2017-11-07:
Hello,

this library didn't work if me using two sensor dht22 on my raspberry pi3 (raspbian, python2). My program only read one sensor. how to resolve it, thanks before.

code: 

import Adafruit_DHT

while 1:

	sensor = Adafruit_DHT.DHT22
 	pin = 13
 	pin2 = 26
	humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
	humidity2, temperature2 = Adafruit_DHT.read_retry(sensor, pin2)

	if humidity != None and temperature != None :
		print('Temp={0:0.1f}*C  Humidity={1:0.1f}%'.format(temperature, humidity))
	if  humidity2 != None and temperature2 != None :
    		print('Temp2={0:0.1f}*C  Humidity2={1:0.1f}%'.format(temperature2, humidity2))
	else:
    		print('Failed to get reading. Try again!')

flyadive, 2017-12-13:
So, Harlilopez, did it work?

sensor = Adafruit_DHT.DHT22
pin = 13
pin2 = 26
humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
humidity2, temperature2 = Adafruit_DHT.read_retry(sensor, pin2)

if humidity != None and temperature != None :
	print('Temp={0:0.1f}*C  Humidity={1:0.1f}%'.format(temperature, humidity))
if  humidity2 != None and temperature2 != None :
		print('Temp2={0:0.1f}*C  Humidity2={1:0.1f}%'.format(temperature2, humidity2))
else:
		print('Failed to get reading. Try again!')

Barry
FLYADIVE@GMAIL.COM

harlilopez, 2017-12-14:
filyadive,

problem was solved, but my raspberry pi can't read data if sensor cable length out of 5 meter.... now i'm use arduino to read sensor dht and communicate it to raspberry pi via data serial...

flyadive, 2017-12-14:
Thank you for the response and update.
My testing will be with leads no longer than 1 meter.
Do you think the problem may be a voltage drop issue?
I’m thinkingm:
1 go to a larger diameter wire.
2 use shielded wire.
3 use the 5 VDC pins instead of the 3 VDC pins.

Ian totally new to the RPi and programming, I’m just considering some basic
electrical engineering.
YES, I am on a huge upward learning curve.
Thank you again,
Barry

rcassel52, 2018-03-02:
I've used DHT22 with 50ft of CAT 5 cable on an Arduino Yun without any issues.   The Yun has 5V pins

brennen, 2018-07-27:
Original issue seems to be resolved.

closed 2018-07-27 by brennen

user2589, 2018-10-10:
@harlilopez ,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/73)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!