adafruit/Adafruit_Python_DHT #74: PyPi out of date?

PeteLawler, 2017-11-20:
Hi!
I realise you only support python2, but I'm trying to fiddle with this on python3 inside a venv. And I can't install from pypi, however I can install from this git repo. Pypi hasn't been updated in a couple of years, I am hoping I may be able to convince you to give it a poke.

Cheers,

Pete.

cerealcable, 2017-12-27:
So the pypi version is from https://github.com/hexdump42/Adafruit_Python_DHT not adafruit, which is confusing at best. I've run into this exact issue and was quite annoyed myself. Not sure if there's something that can be done but it should would be awesome if this was in pypi and up to date.

tedder, 2018-03-18:
see #77

brennen, 2018-07-27:
Current release is live on PyPI.

closed 2018-07-27 by brennen