adafruit/Adafruit_Python_DHT #75: Incorrect board version detected

greenscientist, 2017-12-17:
I have a Raspberry Pi 1 Model B 1.2 and the current code detects it as a Raspberry Pi 2 and then cannot talk to GPIO. 

Here's the content of cpuinfo:
--------------
cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 000d
Serial		: 00000000a2944376
------------
Switching based on the Hardware field seems inaccurate as many Pi 1 use the 2835 platform. 

Maybe just using the revision entry and following 
https://elinux.org/RPi_HardwareHistory#Board_Revision_History
might be a better choice.

rmuller, 2018-02-09:
With newer kernels the reported hardware [is incorrect](https://github.com/raspberrypi/linux/issues/2008).
As suggested by @greenscientist using Revision is the preferred way to do it. Also see https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-obtain_hw_model#L125-L364

mafrosis, 2018-04-17:
I can confirm a very old Pi B+ v1 is reporting itself as a v3 with the BCM2835 chip.

lawin1, 2019-05-25:
Caught the same bug. Solved it by downgrading Raspberry Pi 1 Model B firmware to a version 4.4.50 as were described in this message:

https://github.com/adafruit/Adafruit_Python_DHT/issues/63#issuecomment-285523878

ladyada, 2019-11-08:
hi this repo is going to be deprecated soon, please try 
https://github.com/adafruit/Adafruit_CircuitPython_DHT
which will replace this library

closed 2019-11-08 by ladyada