adafruit/Adafruit_Python_DHT #77: Update Adafruit_Python_DHT library in Pypi

jesusiglesias, 2017-12-27:
Hi,

I have trouble with the execution of a code, which reads data from a DHT11 sensor, when the installation of the library is done with the pip command. My RaspberryPi is a model 3 and I realized that this version is not updated because RPi version 3 is not included in the code when detecting the platform. For this reason, RPi v3 is not recognized and the program returns to the default value (Beaglebone). The transfer of parameters is then of type "String".

Please, the solution is to keep Pypi updated with the latest version of GitHub.

Information:

- Platform/operating system:  **Raspberry Pi with Raspbian**
- Python version:  **python2.7.13**
- Error message you are receiving, including any Python exception traces:  **TypeError: expected string or buffer***
- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **Have a Raspberry Pi v3. Install the library with the pip command. Execute a code where values from DHT11 sensor are read.**

cerealcable, 2017-12-27:
So the pypi version is from https://github.com/hexdump42/Adafruit_Python_DHT not adafruit, which is confusing at best. I've run into this exact issue and was quite annoyed myself. Not sure if there's something that can be done but it should would be awesome if this was in pypi and up to date.

jesusiglesias, 2017-12-27:
This repository [()](https://github.com/hexdump42/Adafruit_Python_DHT) is a fork from original repository but it is outdated and this is the main problem. I think that this one was created to upload this library to Pypi.

ladyada, 2017-12-28:
hiya we opened a ticket https://sourceforge.net/p/pypi/support-requests/789/ hopefully they'll let us take it over! until then please install manually :)

jesusiglesias, 2017-12-28:
Thanks for the info. This is good news and I hope that it will be updated in Pypi soon.

cerealcable, 2017-12-28:
@ladyada Could always package this as Adafruit_DHT and skip Python entirely, it is pypi after all so Python is a little redundant.

FireBelow, 2018-02-12:
This really messed me up a couple of times when I got the two repositories confused.  I really hope you can take control of the Pypi package

ladyada, 2018-02-12:
we'll take a look at that, but it wont stop people from using the old one. we don't even know where people are getting the idea of installing with pip? cause we dont recommend it :/

cerealcable, 2018-02-12:
Honestly, "because python".

It's the first place I almost always look. It makes development projects
avoid describing every dependency and how to install them.

If I hadn't had a background in Python I would agree that it seems so
bizarre. I'll open an issue with the other users GitHub fork to transfer to
you.

In the meantime, it would be so awesome to be able to list adafruit
packages in my requirements.txt files which people then can install with
pip. That would be easiest if there was a new package created while this
situation is resolved.

ladyada, 2018-02-12:
ok that makes more sense! we will add this to our upcoming python library sweeps
if you can help location the author of the current pypi registry it would be great :)

FireBelow, 2018-02-12:
Agree 100%

Sispheor, 2018-03-10:
Haha I got screwed by the pypi libs too.

Guys you should just create another package with a new name and reference it on the readme of this repo.

Anyway, the lib works well. Thanks !

tedder, 2018-03-18:
Since hexdump42 doesn't have a posted email address and github issues are turned off, [I created a pull request to ask them to turn over/share the pypi publishing keys](https://github.com/hexdump42/Adafruit_Python_DHT/pull/2).

PeteLawler, 2018-03-20:
I *think* hexdump42, Mark Rees, might be the same Mark Rees as seen in this corporate bio https://www.corumgroup.com.au/about/management-team - the linkedin profile would seem to match up.

I'll try contacting him today, as he's in my timezone.

hexdump42, 2018-03-21:
Just saw the email with regards this. The reason why I put it up in pypi a couple of years ago was that Adafruit hadn't and for a project I was working on needed to install via pip. If Adafruit want to keep pypi up-to-date I am happy to pass the pypi project across to them.

tedder, 2018-03-21:
That's great! So this is back in @ladyada's hands.

hexdump42, 2018-03-21:
@ladyada Please let me know the pypi user id you want me to assign owner rights to. There were a number of tweaks I made so a pip install worked correctly. These changes may need to be applied to the Adafruit master before building a new package to upload. Happy to provide help if you need it.

ladyada, 2018-03-21:
AHHHHMAZING!!

FireBelow, 2018-03-21:
> @hexdump42 <https://github.com/hexdump42> we are "adafruit-python" if you
> move ownership, i'll figure out how to get this repo set up for pypi and
> 'replace' your code.
> thank you :)

Excellent, thanks everyone!

hexdump42, 2018-03-21:
@ladyada ownership of pypi project transferred

ladyada, 2018-03-21:
thanks a bunch @hexdump42 :)

jesusiglesias, 2018-03-22:
Thanks @ladyada

deanm1278, 2018-05-08:
ok this has been updated! 
https://pypi.org/project/Adafruit_Python_DHT/

closed 2018-05-08 by deanm1278

FireBelow, 2018-05-08:
Great, thanks!

PeteLawler, 2018-05-08:
Excellent community work folks, thanks @hexdump42 @ladyada @deanm1278 !