adafruit/Adafruit_Python_DHT #79: Pi Zero w support

error0327, 2018-01-11:
Hi , 
i was trying to run the code on the Raspberry pi zero w , and even if i can install it with forcing the pi option
since it can't detect what platform is that when trying to use the code i get the following error :

Traceback (most recent call last):
  File "/home/pi/Scripts/Live-Temperature-DHT11/AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 91, in read_retry
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 77, in read
  File "build/bdist.linux-armv6l/egg/Adafruit_DHT/common.py", line 60, in get_platform
RuntimeError: Unknown platform.

so the work around was to copy the lib dir Adafruit_DHT to my script dir , and to made the following change in common.py

def get_platform():
    """Return a DHT platform interface for the currently detected platform."""
    plat = platform_detect.platform_detect()
    plat = platform_detect.RASPBERRY_PI  <<<<<<<<<<<<<
    if plat == platform_detect.RASPBERRY_PI:
        # Check for version 1 or 2 of the pi.
        version = platform_detect.pi_version()
        version = 1 <<<<<<<<<<<<<<<<<
        if version == 1:
            from . import Raspberry_Pi
            return Raspberry_Pi
        elif version == 2:
            from . import Raspberry_Pi_2
            return Raspberry_Pi_2
        else:

the question can you create an official solution for this please , thanks .....

kiote, 2019-06-02:
hi! Were you able to read values with this @error0327? I'm trying the same approach for Nerves + Raspberry pi zero w + DHT11, which uses this driver underneath. It compiles but can't read values due to timeout. 

That's raspberry0 "support" I've added https://github.com/FirstLawofRobotics/nerves_dht/tree/master/src/rpi0

ne0zen, 2019-06-23:
@kiote FYI, I've been able to get readings from a DHT22 with the most recent code on a Raspberry Pi Zero W w/ raspbian stretch.  You should probably try it again.

ladyada, 2019-11-08:
hi this repo is going to be deprecated soon, please try 
https://github.com/adafruit/Adafruit_CircuitPython_DHT
which will replace this library

closed 2019-11-08 by ladyada