adafruit/Adafruit_Python_DHT #81: How to automatic run after reboot the Raspberry Pi with DHT22 sensor

jackyZhangHZ, 2018-01-29:
Hi :-)   i have a question , could you please help: How to run  automatic after reboot the Raspberry Pi with DHT22 sensor
MANY MANY THANKS

ladyada, 2018-01-29:
https://learn.adafruit.com/running-programs-automatically-on-your-tiny-computer

closed 2018-01-29 by ladyada

dade80vr, 2018-07-27:
Hi, use crontab: run `crontab -e` and add this at the end:

`@reboot sleep 30 && python /yourprojectfolder/start.py &`

Leave 30 or 60 sleep seconds to let Rasp to go ready.
Replace "yourprojectfolder" with your.
"&" at end launch project in backgroud

To kill the process, use:

`pkill -ef -9 "yourprojectfolder/start.py"`

Hope this help, bye.

user2589, 2018-10-10:
@jackyZhangHZ ,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/81)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!