adafruit/Adafruit_Python_DHT #83: Can't install on RPI3

FireBelow, 2018-02-09:
I installed using pip then remembered i needed to use pip3 instead.  I uninstalled using pip but when I tried to install using pip3 i get the following errors.  I traced the setup.py file and import and I'm not sure why it is unhappy.

`pi@BeeHiveRaspPi:~ $ pip3 install Adafruit_Python_DHT
Collecting Adafruit_Python_DHT
  Using cached Adafruit_Python_DHT-1.1.2.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-ghg1_42x/Adafruit-Python-DHT/setup.py", line 5, in <module>
        import Adafruit_DHT.platform_detect as platform_detect
      File "/tmp/pip-build-ghg1_42x/Adafruit-Python-DHT/Adafruit_DHT/__init__.py", line 21, in <module>
        from .common import DHT11, DHT22, AM2302, read, read_retry
      File "/tmp/pip-build-ghg1_42x/Adafruit-Python-DHT/Adafruit_DHT/common.py", line 23, in <module>
        import platform_detect
    ImportError: No module named 'platform_detect'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ghg1_42x/Adafruit-Python-DHT/
`

I also get the following similar error

`pi@BeeHiveRaspPi:~/.local/lib/python3.5/site-packages/Adafruit_Python_DHT $ sudo python3 setup.py install
Traceback (most recent call last):
  File "setup.py", line 5, in <module>
    import Adafruit_DHT.platform_detect as platform_detect
  File "/home/pi/.local/lib/python3.5/site-packages/Adafruit_Python_DHT/Adafruit_DHT/__init__.py", line 21, in <module>
    from .common import DHT11, DHT22, AM2302, read, read_retry
  File "/home/pi/.local/lib/python3.5/site-packages/Adafruit_Python_DHT/Adafruit_DHT/common.py", line 23, in <module>
    import platform_detect
ImportError: No module named 'platform_detect'`

FireBelow, 2018-02-10:
I did lots of reading and figured out that I needed to edit the setup.py file to change

`import Adafruit_DHT.platform_detect as platform_detect`

to

`sys.path.append('/home/pi/.local/lib/python3.5/site-packages/Adafruit_Python_DHT/Adafruit_DHT')
from Adafruit_DHT.platform_detect import *`

after that the install worked

`pi@BeeHiveRaspPi:~/.local/lib/python3.5/site-packages/Adafruit_Python_DHT $ sudo python3 setup.py install
running install
running bdist_egg
running egg_info
creating Adafruit_Python_DHT.egg-info
writing top-level names to Adafruit_Python_DHT.egg-info/top_level.txt
writing Adafruit_Python_DHT.egg-info/PKG-INFO
writing dependency_links to Adafruit_Python_DHT.egg-info/dependency_links.txt
writing entry points to Adafruit_Python_DHT.egg-info/entry_points.txt
writing manifest file 'Adafruit_Python_DHT.egg-info/SOURCES.txt'
reading manifest file 'Adafruit_Python_DHT.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Adafruit_Python_DHT.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-3.5
creating build/lib.linux-armv7l-3.5/examples
copying examples/google_spreadsheet.py -> build/lib.linux-armv7l-3.5/examples
copying examples/simpletest.py -> build/lib.linux-armv7l-3.5/examples
copying examples/__init__.py -> build/lib.linux-armv7l-3.5/examples
copying examples/AdafruitDHT.py -> build/lib.linux-armv7l-3.5/examples
creating build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/common.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/platform_detect.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/Raspberry_Pi.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/Beaglebone_Black.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/Test.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/__init__.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
copying Adafruit_DHT/Raspberry_Pi_2.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
running build_ext
building 'Adafruit_DHT.Beaglebone_Black_Driver' extension
creating build/temp.linux-armv7l-3.5
creating build/temp.linux-armv7l-3.5/source
creating build/temp.linux-armv7l-3.5/source/Beaglebone_Black
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c source/_Beaglebone_Black_Driver.c -o build/temp.linux-armv7l-3.5/source/_Beaglebone_Black_Driver.o -std=gnu99
source/_Beaglebone_Black_Driver.c: In function ‘initBeaglebone_Black_Driver’:
source/_Beaglebone_Black_Driver.c:48:5: warning: implicit declaration of function ‘Py_InitModule’ [-Wimplicit-function-declaration]
     Py_InitModule("Beaglebone_Black_Driver", module_methods);
     ^~~~~~~~~~~~~
source/_Beaglebone_Black_Driver.c:49:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c source/common_dht_read.c -o build/temp.linux-armv7l-3.5/source/common_dht_read.o -std=gnu99
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c source/Beaglebone_Black/bbb_dht_read.c -o build/temp.linux-armv7l-3.5/source/Beaglebone_Black/bbb_dht_read.o -std=gnu99
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c source/Beaglebone_Black/bbb_mmio.c -o build/temp.linux-armv7l-3.5/source/Beaglebone_Black/bbb_mmio.o -std=gnu99
source/Beaglebone_Black/bbb_mmio.c: In function ‘bbb_mmio_get_gpio’:
source/Beaglebone_Black/bbb_mmio.c:69:25: warning: argument to ‘sizeof’ in ‘memset’ call is the same expression as the destination; did you mean to dereference it? [-Wsizeof-pointer-memaccess]
   memset(gpio, 0, sizeof(gpio));
                         ^
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.5/source/_Beaglebone_Black_Driver.o build/temp.linux-armv7l-3.5/source/common_dht_read.o build/temp.linux-armv7l-3.5/source/Beaglebone_Black/bbb_dht_read.o build/temp.linux-armv7l-3.5/source/Beaglebone_Black/bbb_mmio.o -lrt -o build/lib.linux-armv7l-3.5/Adafruit_DHT/Beaglebone_Black_Driver.cpython-35m-arm-linux-gnueabihf.so
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/examples
copying build/lib.linux-armv7l-3.5/examples/google_spreadsheet.py -> build/bdist.linux-armv7l/egg/examples
copying build/lib.linux-armv7l-3.5/examples/simpletest.py -> build/bdist.linux-armv7l/egg/examples
[... cut ...]
    print 'usage: sudo ./Adafruit_DHT.py [11|22|2302] GPIOpin#'
                                                              ^
SyntaxError: Missing parentheses in call to 'print'

Adding Adafruit-Python-DHT 1.1.2 to easy-install.pth file
Installing Adafruit_DHT script to /usr/local/bin

Installed /usr/local/lib/python3.5/dist-packages/Adafruit_Python_DHT-1.1.2-py3.5-linux-armv7l.egg
Processing dependencies for Adafruit-Python-DHT==1.1.2
Finished processing dependencies for Adafruit-Python-DHT==1.1.2
`

closed 2018-02-14 by FireBelow

user2589, 2018-10-10:
@FireBelow,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/83)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!