adafruit/Adafruit_Python_DHT #85: Will this work in Ubuntu 16.04 with normal pc

arunnediyasala, 2018-02-26:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**
Hi,
I want to use this code for normal python2.7 running on a Ubuntu 16.04, I am getting the reading of DHT22 sensor as analog value from 0 to 1023 can I able to map to real temperature and humidity.

geofbaum, 2018-02-26:
How are you connecting the sensor to your computer as this library is mainly setup to run on something with an ARM processor you are going to have to change the code some to get it to work for you. Or something that probably would have popped up in a Google search is an example that utilizes an arduino in between the computer and the sensor. Here is the link, [http://forum.arduino.cc/index.php?topic=244703.0](http://forum.arduino.cc/index.php?topic=244703.0) and here is the python code that is shown from that forum post.

```python
#############################################
#    Retrieving the data from a DHT sensor
#    connected to an Arduino Uno board
# by A.LINA June 2014
#############################################

import os
import re
import matplotlib.pyplot as plt
import threading as thr
import serial as srl
import time

ArduinoPort = "/dev/tty.usbmodem1451"
ArduinoSerial = None
DataTime      = None
ArduinoDataH  = None
ArduinoDataT  = None

def InitSerial():
    global ArduinoSerial
    
    ArduinoSerial = srl.Serial(ArduinoPort,
                     baudrate=9600,
                     bytesize=srl.EIGHTBITS,
                     parity=srl.PARITY_NONE,
                     stopbits=srl.STOPBITS_ONE,
                     timeout=2,
                     xonxoff=0,
                     rtscts=0)
    
    time.sleep(2)
    ArduinoSerial.baudrate = 9600
    print("Open serial communication with Arduino")
    print(ArduinoSerial)
    print("======================================")
    return ArduinoSerial.isOpen()

def SetSamplingRate(SamplingRate):
    if(ArduinoSerial.isOpen()):
        ArduinoSerial.readline()
        temp = str(20000+SamplingRate);
        ArduinoSerial.write(temp.encode('utf-8'))
    
def StartSendData():
    if(ArduinoSerial.isOpen()):
        ArduinoSerial.readline()
        temp = str(10001);
        ArduinoSerial.write(temp.encode('utf-8'))

def StopSendData():
    if(ArduinoSerial.isOpen()):
        ArduinoSerial.readline()
        temp = str(10000);
        ArduinoSerial.write(temp.encode('utf-8'))

def GetData(DurationSec):
    global ArduinoDataH, ArduinoDataT, DataTime

    DataTime=[]
    ArduinoDataH=[]
    ArduinoDataT=[]
    Inc=0

    InitTime=time.time()
    
    # dummy read to flush the port
    ArduinoSerial.readline()

    while((time.time()-InitTime)<DurationSec):
        print str(Inc)+" "+str(ArduinoSerial.isOpen())+" "+str(time.time()-InitTime)
        data=str(ArduinoSerial.readline())
        parse_data = re.match(r"DHT:H ([-]{0,1}\d+) T ([-]{0,1}\d+)", data)
        if(parse_data!=None):
            if(len(parse_data.groups())==2):
                print "ADD"
                Humidity    = int(parse_data.group(1))
                Temperature = int(parse_data.group(2))
                DataTime.append(time.time()-InitTime)
                ArduinoDataH.append(float(Humidity/1024.0))
                ArduinoDataT.append(float(Temperature/1024.0))
        Inc+=1
        
    print "END:"+str(ArduinoSerial.isOpen())+" "+str(time.time()-InitTime)

def PlotData():
    plt.title("Arduino DHT data")

    plt.subplot(2,1,1, axisbg='y')
    plt.plot(DataTime[1:], ArduinoDataT[1:], 'r-', DataTime[1:], ArduinoDataT[1:], 'go')
    plt.xlabel('time (iter)')
    plt.ylabel('Temperature')

    plt.subplot(2,1,2, axisbg='y')
    plt.plot(DataTime[1:], ArduinoDataH[1:], 'r-', DataTime[1:], ArduinoDataH[1:], 'go')
    plt.xlabel('time (iter)')
    plt.ylabel('Humidity')
    
    plt.show(block=False)

def MainArduino(SamplingRate, DurationSec):
    if(InitSerial()==True):
        time.sleep(3)
        SetSamplingRate(SamplingRate)
        time.sleep(3)
        StartSendData()
        time.sleep(3)
        GetData(DurationSec)
        time.sleep(3)
        StopSendData()
        time.sleep(3)
        PlotData()
    ArduinoSerial.close()
```

It might point you in the right direction since if you're getting values from 0 to 1023 already all you need to do is convert it which doesn't require this library, just a few more lines of code.

arunnediyasala, 2018-02-26:
I have XBee device which is connected to sensor and I am controlling XBee through my computer

geofbaum, 2018-02-26:
@arunnediyasala 
Ok well that's a start though since you are already receiving a value from the ADC on the Xbee. The problem or the point here is that you should be receiving the equivalent of 5 segments of 8-bits for each packet from the DHT sensor and you are receiving responses in 10-bits (0-1023). Are you able to get the raw hex value of the 10-bit reponses? If so you could possibly get the most significant 8-bit and from there get your humidity and temperature values.

Remembering the segment order of the DHT response starts with the Humidity integral  value followed by Humidity decimal value, then the Temperature integral value followed by the Temperature decimal value and finally the checksum value.

arunnediyasala, 2018-02-26:
@geofbaum Thank you for your replay. I can only able to get the integer value range from 0 to 1023. I will also check is there is any way to get value as binary if then I will extract the humidity and temperature .

geofbaum, 2018-02-26:
Well 1023 is just 0b1111111111 in binary so if anything that is just a simple line in python to convert it. But the reason I asked was just from my experience with Xbees when I was checking out data being sent it always was in hex format. It is just hard to know what values you are seeing to better advise.

arunnediyasala, 2018-02-26:
@geofbaum I am using programmable XBee and we can read the ADC pin value it is in the form of integer. There is a python program which will read each pin information that is also in pure integer.So here I am going to extract the data fields before that I want to know the data filed of DHT11 , then I can extract data for temperature and humidity.

brennen, 2018-07-31:
This seems... Basically resolved?

closed 2018-07-31 by brennen

user2589, 2018-10-10:
@arunnediyasala,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/85)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!