adafruit/Adafruit_Python_DHT #87: DHT22 humidity and temperature values are incorrect

preranagawale, 2018-03-19:
I'm using R-pi 3 with Python 2.7.13 and the DHT22 sensor for reading humidity and temperature values. I tried changing surrounding temperature but it does not read accurate values. In fact, it does not show significant change. I tried changing sensor but still facing the same issue. While when I use the same sensor with ESP8266 it shows a significant change in values.
Please suggest solutions.

closed 2018-03-19 by preranagawale

geofbaum, 2018-04-08:
@preranagawale So why did you close this issue? It might help others who are seeing the same thing that you did if you described what it was that fixed your problem.

preranagawale, 2018-04-09:
The problem was with code. I was reading sensor values at the start and printing it in the loop. So basically It used to read sensor only once in some defined time interval and that's why it was unable to show significant changes.

user2589, 2018-10-10:
@preranagawale,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/87)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!