adafruit/Adafruit_Python_DHT #89: setup.py fails to install with error "TypeError: __init__() takes exactly 2 arguments (4 given)"

DoctorOctagonapus, 2018-03-28:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry Pi 2 running Raspbian Stretch

- Python version (run `python -version` or `python3 -version`):  python --version: 2.7.13; python3 --version: 3.5.3

- Error message you are receiving, including any Python exception traces: 
Output from "sudo python setup.py install":
`running install`
`running bdist_egg`
`running egg_info`
`writing Adafruit_DHT.egg-info/PKG-INFO`
`writing top-level names to Adafruit_DHT.egg-info/top_level.txt`
`writing dependency_links to Adafruit_DHT.egg-info/dependency_links.txt`
`reading manifest file 'Adafruit_DHT.egg-info/SOURCES.txt'`
`writing manifest file 'Adafruit_DHT.egg-info/SOURCES.txt'`
`installing library code to build/bdist.linux-armv7l/egg`
`running install_lib`
`running build_py`
`running build_ext`
`creating build/bdist.linux-armv7l/egg`
`creating build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/Beaglebone_Black.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/common.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/__init__.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/platform_detect.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/Test.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/Raspberry_Pi.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/Raspberry_Pi_2.py -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`copying build/lib.linux-armv7l-2.7/Adafruit_DHT/Raspberry_Pi_2_Driver.so -> build/bdist.linux-armv7l/egg/Adafruit_DHT`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/Beaglebone_Black.py to Beaglebone_Black.pyc`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/common.py to common.pyc`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/__init__.py to __init__.pyc`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/platform_detect.py to platform_detect.pyc`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/Test.py to Test.pyc`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/Raspberry_Pi.py to Raspberry_Pi.pyc`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/Raspberry_Pi_2.py to Raspberry_Pi_2.pyc`
`creating stub loader for Adafruit_DHT/Raspberry_Pi_2_Driver.so`
`byte-compiling build/bdist.linux-armv7l/egg/Adafruit_DHT/Raspberry_Pi_2_Driver.py to Raspberry_Pi_2_Driver.pyc`
`creating build/bdist.linux-armv7l/egg/EGG-INFO`
`copying Adafruit_DHT.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO`
`copying Adafruit_DHT.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO`
`copying Adafruit_DHT.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO`
`copying Adafruit_DHT.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO`
`writing build/bdist.linux-armv7l/egg/EGG-INFO/native_libs.txt`
`zip_safe flag not set; analyzing archive contents...`
`creating 'dist/Adafruit_DHT-1.3.2-py2.7-linux-armv7l.egg' and adding 'build/bdist.linux-armv7l/egg' to it`
`removing 'build/bdist.linux-armv7l/egg' (and everything under it)`
`Processing Adafruit_DHT-1.3.2-py2.7-linux-armv7l.egg`
`Removing /usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.3.2-py2.7-linux-armv7l.egg`
`Copying Adafruit_DHT-1.3.2-py2.7-linux-armv7l.egg to /usr/local/lib/python2.7/dist-packages`
`Adafruit-DHT 1.3.2 is already the active version in easy-install.pth`

`Installed /usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.3.2-py2.7-linux-armv7l.egg`
`Traceback (most recent call last):`
  `File "setup.py", line 94, in <module>`
    `ext_modules       = extensions)`
  `File "/usr/lib/python2.7/distutils/core.py", line 151, in setup`
    `dist.run_commands()`
  `File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands`
    `self.run_command(cmd)`
  `File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command`
    `cmd_obj.run()`
  `File "build/bdist.linux-armv7l/egg/setuptools/command/install.py", line 65, in run`
  `File "build/bdist.linux-armv7l/egg/setuptools/command/install.py", line 115, in do_egg_install`
  `File "build/bdist.linux-armv7l/egg/setuptools/command/easy_install.py", line 360, in run`
  `File "build/bdist.linux-armv7l/egg/setuptools/command/easy_install.py", line 576, in easy_install`
  `File "build/bdist.linux-armv7l/egg/setuptools/command/easy_install.py", line 627, in install_item`
  `File "build/bdist.linux-armv7l/egg/setuptools/command/easy_install.py", line 677, in process_distribution`
`TypeError: __init__() takes exactly 2 arguments (4 given)`

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):
`git clone https://github.com/adafruit/adafruit_python_dht.git`
`cd adafruit_python_dht`
`sudo apt install build-essential python-dev`
`sudo python setup.py install`

Any thoughts on what might be causing this? I previously had it working but had to reinstall.

geofbaum, 2018-03-29:
This is probably not a specific issue with this library as I've seen this issue pop up elsewhere but you might want to try to update your distutils and setuptools. I don't know if that will fix it but it might be a start.

DoctorOctagonapus, 2018-05-01:
Hey, I was never able to get it working and the distutils/setuptools were up-to-date at the time. Not tried it since as my HomeAssistant instance was able to pick up and query the sensor with no issues. Will try again tonight.

user2589, 2018-10-10:
@DoctorOctagonapus,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/89)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!