adafruit/Adafruit_Python_DHT #91: I can not install RaspBerry 3 and Raspbian Jessie Lite

cusna, 2018-04-26:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry Raspbian Jessie lite

- Python version (run `python -version` or `python3 -version`):  2.7

- Error message you are receiving, including any Python exception traces:   File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): root@raspberrypi:~/Adafruit_Python_DHT# sudo python setup.py install
Extracting in /tmp/tmp1VV7R9
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    use_setuptools()
  File "/root/Adafruit_Python_DHT/ez_setup.py", line 140, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/root/Adafruit_Python_DHT/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/root/Adafruit_Python_DHT/ez_setup.py", line 62, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/root/Adafruit_Python_DHT/ez_setup.py", line 100, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/root/Adafruit_Python_DHT/ez_setup.py", line 88, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file
root@raspberrypi:~/Adafruit_Python_DHT#

geofbaum, 2018-04-26:
Can you try to redownload the files and try again? Also have you made sure your python environment is up to date? And lastly can you describe to me how you downloaded the file?  This error is possibly due to a corrupt file that was downloaded. Haven't had this issue myself working in Python 2.7 or 3 on a Pi3 which is why I ask.

cusna, 2018-04-27:
I'll show you all the steps I do:
`root@raspberrypi:/home/pi# cd /
root@raspberrypi:/# clear
root@raspberrypi:/# cd /home/pi
root@raspberrypi:/home/pi# git clone https://github.com/adafruit/Adafruit_Python_DHT.git
Cloning into 'Adafruit_Python_DHT'...
remote: Counting objects: 249, done.
remote: Total 249 (delta 0), reused 0 (delta 0), pack-reused 249
Ricezione degli oggetti: 100% (249/249), 77.00 KiB | 0 bytes/s, done.
Risoluzione dei delta: 100% (142/142), done.
Checking connectivity... fatto.
root@raspberrypi:/home/pi# cd Adafruit_Python_DHT/
root@raspberrypi:/home/pi/Adafruit_Python_DHT# sudo apt-get install build-essential python-dev
Lettura elenco dei pacchetti... Fatto
Generazione albero delle dipendenze
Lettura informazioni sullo stato... Fatto
build-essential è già alla versione più recente.
python-dev è già alla versione più recente.
0 aggiornati, 0 installati, 0 da rimuovere e 3 non aggiornati.
root@raspberrypi:/home/pi/Adafruit_Python_DHT# python setup.py build
Downloading https://pypi.python.org/packages/source/s/setuptools/setuptools-4.0.1.zip
Extracting in /tmp/tmpjAJfvJ
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 140, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 62, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 100, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 88, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file
root@raspberrypi:/home/pi/Adafruit_Python_DHT# sudo python setup.py install
Extracting in /tmp/tmpWG7D4E
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 140, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 62, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 100, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 88, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 811, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file
root@raspberrypi:/home/pi/Adafruit_Python_DHT#

Thanks`

geofbaum, 2018-04-27:
@cusna 

This is what I see when I do the install.
```bash
pi@raspberrypi:~ $ git clone https://github.com/adafruit/Adafruit_Python_DHT.gitCloning into 'Adafruit_Python_DHT'...
remote: Counting objects: 249, done.
remote: Total 249 (delta 0), reused 0 (delta 0), pack-reused 249
Receiving objects: 100% (249/249), 77.00 KiB | 0 bytes/s, done.
Resolving deltas: 100% (142/142), done.
```
Followed by:
```bash
pi@raspberrypi:~/Adafruit_Python_DHT $ sudo apt-get install build-essential python-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.3).
python-dev is already the newest version (2.7.13-2).
python-dev set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 13 not upgraded.
```

Is there a reason you did this step instead of simply just running python setup.py install?
```bash
root@raspberrypi:/home/pi/Adafruit_Python_DHT# python setup.py build
```
I ask because you get the same error when trying to do that and it again seems like there's a corrupted file somewhere. Maybe it's an issue with the Lite version of the OS install as you shouldn't need to download a copy of setuptools if everything is setup properly with your python installation.

Maybe try
```bash
python -m pip install --upgrade pip setuptools wheel
```
Before running python setup.py install.

sim51, 2018-04-29:
I had the same problem today.

As a workaround, I just download `setuptools-4.0.1.zip` manually in the folder, and relaunch the install.
I don't know why the downloaded zip file from the installer is not good, perhaps my downloader was not configure to follow redirections

geofbaum, 2018-04-29:
@sim51 Glad that worked for you. The thing that's odd to me though is you shouldn't need to install from that zip file if you already have setuptools in your Python environment which I personally didn't think was missing from my last Raspbian Lite install's Python env. but maybe I added it after the fact.

cusna, 2018-05-02:
Hello,
I also solved by manually downloading the file `setuptools-4.0.1.zip`
thanks

phgachoud, 2018-05-09:
Comment from https://github.com/adafruit/Adafruit_Python_DHT/issues/92 @bargacha  solved my problem...

> Delete setuptools-4.0.1.zip
> 
> and try
> 
> wget https://files.pythonhosted.org/packages/cb/e0/b5ecf534e14304e62c781614d683d0daa571efe92678901559ea84c1559e/setuptools-4.0.1.zip
> 
> More info here : https://pypi.org/project/setuptools/
> Choose "release history", and look for 4.0.1 (Old version released in 2015)

pepoweb, 2018-06-07:
I can confirm the install crashes on the python setu.py install.

The Ansible playbook I wrote for the install last spring has worked good until about 6 weeks ago. Now I have the RPi's in, it is a bit annoying the time I invested in creating the playbook isn't payed back by a smooth install of all the RPi's at the same time. Oh, well, at least I learned the basics on Ansible :-)

closed 2018-07-31 by brennen

user2589, 2018-10-10:
@cusna,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: [link](https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/91)

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!

commit 05c3cfe3 (2018-07-31): remove deprecated ez_setup.py; update install instructions (fixes #91)
commit 18846dee (2018-07-31): Merge pull request #101 from adafruit/remove_ez_setup

remove deprecated ez_setup.py; update install instructions (fixes #91)