adafruit/Adafruit_Python_DHT #92: Broken link in ez_setup

terminal3, 2018-05-01:
setuptools is no longer found at https://pypi.python.org/packages/source/s/setuptools/

bargacha, 2018-05-05:
Same issue :/

There is something similar here #44 But even with the add zip, it does not work.

bargacha, 2018-05-05:
Delete setuptools-4.0.1.zip

and try

`wget https://files.pythonhosted.org/packages/cb/e0/b5ecf534e14304e62c781614d683d0daa571efe92678901559ea84c1559e/setuptools-4.0.1.zip`

Tronde, 2018-05-19:
Thank you for your [suggestion](https://github.com/adafruit/Adafruit_Python_DHT/issues/92#issuecomment-386798942). I had downloaded setuptools from the link you mentioned and was able to complete the installation.

The [Readme.md](https://github.com/adafruit/Adafruit_Python_DHT/blob/master/README.md) says that one do not need to use sudo when installing on Raspberry Pi. But in my case using Raspbian GNU/Linux 9 (stretch) the installation failed when omitting sudo.

brennen, 2018-07-27:
Duplicates #91.

closed 2018-07-27 by brennen