adafruit/Adafruit_Python_DHT #93: AdafruitDHT.py show higher temp

deffrak, 2018-05-01:
Hi At All,

Today I found a very difficult issue. I make some tests with DHT22 and DHT11 and the result I get is wrong.
For example:

sudo python AdafruitDHT.py 22 4
Temp=563.2*  Humidity=947.2%

sudo python2 AdafruitDHT.py 22 4
Temp=563.2*  Humidity=947.2%

sudo python2.7 AdafruitDHT.py 22 4
Temp=563.2*  Humidity=947.2%

The output file show follow code:
`print('Temp={0:0.1f}*  Humidity={1:0.1f}%'.format(temperature, humidity))`

Any Idea why the DHT22 show higher results?

deffrak, 2018-05-01:
Ok, that happens if you try to call the wrong typ of DHT. 

I make follow change 
`sudo python2.7 AdafruitDHT.py 22 4` to `sudo python2.7 AdafruitDHT.py 11 4`

geofbaum, 2018-05-01:
@deffrak Can you please close this issue now that you've figured out the mistake.

user2589, 2018-10-09:
@deffrak ,

I couldn’t find a better way to contact you so I’m writing here. This comment will be deleted in one week to keep the issue discussion clean.

My name is Marat, I am a PhD student at Carnegie Mellon University. As a part of my research, I’m trying to better understand the sources of issue reports in open source.

Can you please help me by answering few questions about this issue? It shouldn’t take more than a couple of minutes. Here’s the google form: 
https://docs.google.com/forms/d/e/1FAIpQLSfTlhMeh25AZAr0wy-DNBki_HzFP_fqdT3eIk1mf7b80mg8Lg/viewform?usp=pp_url&entry.2074867396=https://github.com/adafruit/Adafruit_Python_DHT/issues/93

I will share everything I learn from answers to this survey with the community afterwards.

Thank you very much!