adafruit/Adafruit_Python_DHT #96: Problem witch installing

Piotr-Pht-Jasiek, 2018-05-16:
I read previous entries and did not find a solution.

pi@raspberrypi:~/Adafruit_Python_DHT $ sudo python setup.py install
Extracting in /tmp/tmpyBCpQu
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    use_setuptools()
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 145, in use_setuptools
    return _do_download(version, download_base, to_dir, download_delay)
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 120, in _do_download
    _build_egg(egg, archive, to_dir)
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 62, in _build_egg
    with archive_context(archive_filename):
  File "/usr/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 100, in archive_context
    with ContextualZipFile(filename) as archive:
  File "/home/pi/Adafruit_Python_DHT/ez_setup.py", line 88, in __new__
    return zipfile.ZipFile(*args, **kwargs)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
zipfile.BadZipfile: File is not a zip file

I use rpi 1 B version

geofbaum, 2018-05-16:
This is the exact same issue that has been posted in other issues. (#91 and #92 ) If the work arounds to download the zip file didn't work you can just install setuptools which I mentioned in one of those two issues. Here is my suggestion again which you should do in a terminal window.
````bash
python -m pip install --upgrade pip setuptools wheel
````
If you get an error from that please let me know but if you don't get an error then you can try to do your install again.

Piotr-Pht-Jasiek, 2018-05-22:
@geofbaum I have a different problem now but not a library. Can you contact me by email? piotr.jasiek@s-m-s.pl

geofbaum, 2018-05-22:
Yes sure no problem, I will send you an email in a minute or two. Sorry for
the delayed response.

brennen, 2018-07-27:
Duplicates #91.

closed 2018-07-27 by brennen