adafruit/Adafruit_Python_DHT #99: Failed pip Install of Adafruit_Python_DHT on Raspbian Stretch

TheCedarPrince, 2018-06-26:
Hey all,

I am having an issue installing the Adafruit Python DHT library via pip3 on the Raspbian Stretch (specs below). 

Basically, I am trying to install it via pip and I am getting a compilation error at "arm-linux-gnueabihf-gcc" saying that some drivers, which should have been built during install, are not there.

I was able to successfully install this from a direct install (e.g. clone the repository and run 'sudo python3 set_up.py install'). My suspicion is something is up with the build-essential package, but I have updated that package as well as python-dev per Adafruit's recommendations. 

Any thoughts of what could be going on? 

~ TheCedarPrince

**Specs:**

- **Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):** Raspberry Pi 3 B with Raspbian Stretch

- **Python version (run `python -version` or `python3 -version`):**  python3.5.3

- **Error message you are receiving, including any Python exception traces:**  

The main suspect lines are these: 

'arm-linux-gnueabihf-gcc: error: source/_Raspberry_Pi_2_Driver.c: No such file or directory'
'arm-linux-gnueabihf-gcc: fatal error: no input files compilation terminated.'

The rest of the dump is here:

https://pastebin.com/9CazaEDk

- **List the steps to reproduce the problem below (if possible attach code or commands
  to run):** 

I executed the following on my RPi:

'sudo apt-get install build-essential'
'sudo apt-get install python-dev'
'pip3 install Adafruit_Python_DHT'

I also tried the following pip commands to no avail:

'pip install Adafruit_Python_DHT'
'sudo pip3 install Adafruit_Python_DHT'
'sudo pip install Adafruit_Python_DHT'

trickv, 2018-07-24:
Same issue here, on Raspbian Jessie.  It looks to me like the latest package pushed to pypi isn't complete; it doesn't contain any of the .c source files which are causing the compile to fail.
`
[trick@radiopi ~]$ sudo pip3 install Adafruit_Python_DHT
Collecting Adafruit_Python_DHT
  Using cached https://files.pythonhosted.org/packages/52/aa/418b82e36fc57316137679c2a12b8bae89ab73f211cbca91a3275e0b5fc2/Adafruit_Python_DHT-1.3.2.tar.gz                                                          
Building wheels for collected packages: Adafruit-Python-DHT                                                                                                                                                         
  Running setup.py bdist_wheel for Adafruit-Python-DHT ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-h9ul_fqm/Adafruit-Python-DHT/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace(
'\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp2xj1osh9pip-wheel- --python-tag cp35:                                                                                         
  running bdist_wheel
  running build
  running build_py
  creating build                                                                                                                                                                                                    
  creating build/lib.linux-armv7l-3.5                                                                                                                                                                               
  creating build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/Test.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/Raspberry_Pi_2.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/Beaglebone_Black.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/common.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/platform_detect.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/Raspberry_Pi.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  copying Adafruit_DHT/__init__.py -> build/lib.linux-armv7l-3.5/Adafruit_DHT
  running build_ext
  building 'Adafruit_DHT.Raspberry_Pi_2_Driver' extension
  creating build/temp.linux-armv7l-3.5
  creating build/temp.linux-armv7l-3.5/source
  creating build/temp.linux-armv7l-3.5/source/Raspberry_Pi_2
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-RUbMX3/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate
-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c source/_Raspberry_Pi_2_Driver.c -o build/temp.linux-armv7l-3.5/source/_Raspberry_Pi_2_Driver.o -std=gnu99
  arm-linux-gnueabihf-gcc: error: source/_Raspberry_Pi_2_Driver.c: No such file or directory
  arm-linux-gnueabihf-gcc: fatal error: no input files
  compilation terminated.
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
`
The published tar archive doesn't even contain the .c files:
`
[trick@radiopi ~/tmp/a]$ curl -s https://files.pythonhosted.org/packages/52/aa/418b82e36fc57316137679c2a12b8bae89ab73f211cbca91a3275e0b5fc2/Adafruit_Python_DHT-1.3.2.tar.gz  | tar xfvz -
Adafruit_Python_DHT-1.3.2/
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Beaglebone_Black.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/common.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/platform_detect.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Raspberry_Pi.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Raspberry_Pi_2.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Test.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/__init__.py
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/dependency_links.txt
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/PKG-INFO
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/SOURCES.txt
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/top_level.txt
Adafruit_Python_DHT-1.3.2/PKG-INFO
Adafruit_Python_DHT-1.3.2/setup.cfg
Adafruit_Python_DHT-1.3.2/setup.py
Adafruit_Python_DHT-1.3.2/source/
Adafruit_Python_DHT-1.3.2/source/Test/
Adafruit_Python_DHT-1.3.2/source/Test/test_dht_read.c
Adafruit_Python_DHT-1.3.2/source/_Test_Driver.c
`

I think the package build process isn't right.

dade80vr, 2018-07-25:
+1 same issue here

brennen, 2018-07-25:
Good call, @trickv. Will get a version with included *.c up shortly.

pptp-plg, 2018-07-26:
"pip3 install Adafruit_Python_DHT" Erorr
can't install from https://pypi.org/project/Adafruit_Python_DHT

brennen, 2018-07-26:
Should be fixed with merge of #67.  v1.3.3 now uploaded to PyPI.

closed 2018-07-26 by brennen

pptp-plg, 2018-07-27:
in https://pypi.org/project/Adafruit_Python_DHT show 1.3.2 lasted version

nhorvath, 2018-07-27:
@brennen It's not showing in pypi.

geofbaum, 2018-07-27:
That's because I believe he updated this one which was the one they were using before they got the access rights that are mentioned here (#77).

The other one is updated to 1.3.3:
https://pypi.org/project/Adafruit-DHT/

pptp-plg, 2018-07-27:
please update  Adafruit_Python_DHT too ,for home assistant

nhorvath, 2018-07-27:
@geofbaum does that mean the package in home assistant should be changed to Adafruit-DHT?

Because this definitely doesn't work:
```
pip install Adafruit_Python_DHT==1.3.3
Collecting Adafruit_Python_DHT==1.3.3
  Could not find a version that satisfies the requirement Adafruit_Python_DHT==1.3.3 (from versions: 1.1.0, 1.1.1, 1.1.2, 1.3.2)
No matching distribution found for Adafruit_Python_DHT==1.3.3
```

brennen, 2018-07-27:
Hey everybody.  Sorry, yeah, I missed that we were evidently updating... Both PyPI packages, I guess?  I'm not actually sure what's going on there, but I'll get it sorted out.

In the meanwhile, [Adafruit_Python_DHT](https://pypi.org/project/Adafruit_Python_DHT/) has been manually updated to 1.3.3.

brennen, 2018-07-31:
@superpuffin we'll keep both of those updated for any changes this library receives.

That said, I think we're unlikely to put much more work into this library.  In the general case, we're shifting our efforts to focus on CircuitPython-compatible libraries, and so probably deprecating this in favor of [Adafruit_CircuitPython_DHT](https://github.com/adafruit/Adafruit_CircuitPython_DHT).

TheCedarPrince, 2018-07-31:
Good to know - the CircuitFruit API looks dead simple and well-documented. Would it be wise to start migrating projects utilizing DHT11 and DHT22 sensors to that platform @brennen ?

ladyada, 2018-07-31:
the DHT sensor is particularly weird, so we're not sure if/when it'll get ported to raspi circuitpython

that said the I2C sensors and SPI devices should all work now, for stuff like BME280 and other I2C device - please try those, we'll be migrating to only supporting circuitpython so we dont double the work  :)

https://learn.adafruit.com/circuitpython-on-raspberrypi-linux