adafruit/Adafruit_Python_DHT #102: pip package looks broken

Vanuan, 2018-08-01:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  alpine on armhf
- Python version (run `python -version` or `python3 -version`):  3.6.6
- Error message you are receiving, including any Python exception traces:  

```
pip3 install Adafruit_Python_DHT==1.3.2
Collecting Adafruit_Python_DHT==1.3.2
  Using cached https://files.pythonhosted.org/packages/52/aa/418b82e36fc57316137679c2a12b8bae89ab73f211cbca91a3275e0b5fc2/Adafruit_Python_DHT-1.3.2.tar.gz
Building wheels for collected packages: Adafruit-Python-DHT
	  Running setup.py bdist_wheel for Adafruit-Python-DHT ... error
  Complete output from command /usr/local/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rqhatc54/Adafruit-Python-DHT/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-8moo75hn --python-tag cp36:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-3.6
  creating build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/__init__.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/Beaglebone_Black.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/platform_detect.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/Raspberry_Pi_2.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/common.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/Raspberry_Pi.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  copying Adafruit_DHT/Test.py -> build/lib.linux-armv7l-3.6/Adafruit_DHT
  running build_ext
  building 'Adafruit_DHT.Raspberry_Pi_2_Driver' extension
  creating build/temp.linux-armv7l-3.6
  creating build/temp.linux-armv7l-3.6/source
  creating build/temp.linux-armv7l-3.6/source/Raspberry_Pi_2
  gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -DTHREAD_STACK_SIZE=0x100000 -fPIC -I/usr/local/include/python3.6m -c source/_Raspberry_Pi_2_Driver.c -o build/temp.linux-armv7l-3.6/source/_Raspberry_Pi_2_Driver.o -std=gnu99
  gcc: error: source/_Raspberry_Pi_2_Driver.c: No such file or directory
  gcc: fatal error: no input files
  compilation terminated.
  error: command 'gcc' failed with exit status 1

```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): `pip3 install Adafruit_Python_DHT==1.3.2`

So I've downloaded a package under this link https://files.pythonhosted.org/packages/52/aa/418b82e36fc57316137679c2a12b8bae89ab73f211cbca91a3275e0b5fc2/Adafruit_Python_DHT-1.3.2.tar.gz
and indeed there's no source code for raspberry pi driver. Here's a contents:

```
Adafruit_Python_DHT-1.3.2/
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Beaglebone_Black.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/common.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/platform_detect.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Raspberry_Pi.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Raspberry_Pi_2.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/Test.py
Adafruit_Python_DHT-1.3.2/Adafruit_DHT/__init__.py
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/dependency_links.txt
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/PKG-INFO
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/SOURCES.txt
Adafruit_Python_DHT-1.3.2/Adafruit_Python_DHT.egg-info/top_level.txt
Adafruit_Python_DHT-1.3.2/PKG-INFO
Adafruit_Python_DHT-1.3.2/setup.cfg
Adafruit_Python_DHT-1.3.2/setup.py
Adafruit_Python_DHT-1.3.2/source/
Adafruit_Python_DHT-1.3.2/source/Test/
Adafruit_Python_DHT-1.3.2/source/Test/test_dht_read.c
Adafruit_Python_DHT-1.3.2/source/_Test_Driver.c
```

How is it supposed to work?

Vanuan, 2018-08-01:
Ok, versions 1.3.3 and 1.3.4 do include sources. But the were released only 4 days ago. I can't imagine nobody experienced this issue. Was 1.3.2 version of package replaced or something?

geofbaum, 2018-08-01:
@Vanuan Take a look at the closed issues. There were about 20+ of them the other day that got closed because of the updates that were pushed and you should see you weren't the only one who had an issue.

Vanuan, 2018-08-02:
Ok, found one: https://github.com/adafruit/Adafruit_Python_DHT/issues/99

Still doesn't explain how did it happen and why there are now 2 pip packages published:
https://pypi.org/project/Adafruit-DHT
https://pypi.org/project/Adafruit_Python_DHT/

:confused:

closed 2018-08-02 by Vanuan

geofbaum, 2018-08-02:
@Vanuan Because the Adafruit_Python_DHT package wasn't controlled by Adafruit until recently. It was initially loaded up by someone else and Adafruit had uploaded the Adafruit-DHT package as their official package (please see #77 ).