adafruit/Adafruit_Python_DHT #103: Installation failure on Windows 10

kkotowick, 2018-08-05:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Windows 10 Enterprise, build 16299

- Python version (run `python -version` or `python3 -version`):  3.7 64-bit

- Error message you are receiving, including any Python exception traces:  

> Collecting Adafruit_DHT
>   Using cached https://files.pythonhosted.org/packages/7f/04/faf8ffa98f2ad8a8fe0f87ffa64690460f6f95a30a25385ea0a95263bf56/Adafruit_DHT-1.3.4.tar.gz
>     Complete output from command python setup.py egg_info:
>     Traceback (most recent call last):
>       File "<string>", line 1, in <module>
>       File "C:\Users\username\AppData\Local\Temp\pip-install-mjpysq9o\Adafruit-DHT\setup.py", line 49, in <module>
>         platform = platform_detect.platform_detect()
>       File "C:\Users\username\AppData\Local\Temp\pip-install-mjpysq9o\Adafruit-DHT\Adafruit_DHT\platform_detect.py", line 40, in platform_detect
>         pi = pi_version()
>       File "C:\Users\username\AppData\Local\Temp\pip-install-mjpysq9o\Adafruit-DHT\Adafruit_DHT\platform_detect.py", line 90, in pi_version
>         with open('/proc/cpuinfo', 'r') as infile:
>     FileNotFoundError: [Errno 2] No such file or directory: '/proc/cpuinfo'
>     ----------------------------------------
> Command "python setup.py egg_info" failed with error code 1 in C:\Users\username\AppData\Local\Temp\pip-install-mjpysq9o\Adafruit-DHT\

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
> pip3 install Adafruit_DHT

It's clearly trying to find a Linux file on a Windows machine, which has obvious issues. Is there any way to get this library working with Python on Windows?

geofbaum, 2018-08-05:
@kkotowick I believe the answer to your first question is, no as this library was designed for devices that for the most part are running some sort of Linux environment as their OS. So I would suspect a large part of the library would have to be rewritten. As for the the a DHT sensor and Windows 10 IoT, the only examples that I've seen are in C++ which is then pretty much a port or copy of what was written for the Arduino type devices.

brennen, 2018-08-09:
Yeah, this library is explicitly not supported on Windows.  That said, there's some chance you could get it to _install_, if not actually run, under the Windows Subsystem for Linux.

closed 2018-08-09 by brennen