adafruit/Adafruit_Python_DHT #110: Cable Length Causes incorrect DHT11 Readings

Brets0150, 2018-12-10:
Hello,
This is an old issue that seems to be back. With a longer cable and a DHT11 sensor, the readings will have too low of a temp and too high of a humidity reading. 

- Platform/operating : Raspberry 3 rev B / Raspbian(Debian 9) 
- Python version: Python 2.7.9 
- Adafruit_Python_DHT v1.4.0
- Error: NA, Giving too low of Temp and too high humidity readings.

To fix the issue edit line 75 of "source/Raspberry_Pi_2/pi_2_dht_read.c".
Change  From: "for (volatile int i = 0; i < 50; ++i) {"
TO:  "for (volatile int i = 0; i < 500; ++i) {"

Then rerun the setup.py scripts. The issue should be resolved now.

lendsle, 2019-02-03:
Hi there,

I ran into an issue while trying to readout a DHT22 over a 20 m cable.
The readout wasn't successful even after I just connected a 10 m cable in parallel. The sensor was still directly connected on the breadboard -> raspberry pi (same length about 10-20 cm that used to work before). Seems like the long cable introduced some noise that blocked the readout of the sensor.

I had a look at the code and added a debouncer to reduce noise/false positives.
After that I was able to readout a DHT22 over a 20 m cable - powered only by the 3.3 V of the Raspberry pin #1 - without an additional pull up resistor (a 4.7k seems to be build in at sensor side out of the box - at least this is what I measured between VCC and DATA with a multimeter).

Here is what I've changed to make it run for me:
https://github.com/lendsle/Adafruit_Python_DHT/tree/add_debouncer_to_make_readout_more_robust_for_longer_cable_lengths

@tdicola if this is interesting/useful for you - feel free to pull it :sunglasses:

rmaes4, 2019-06-21:
Thank you for this! My humidity readings on my AM2302 just went from 3,300% to around 20%

lendsle, 2019-06-22:
Thanks for your feedback. Actually I was experimenting with my DHT11 at a new location the day before yesterday because the readout was not stable enough: like 1 of 20 was ok.

Actually I modified the readout further and saved the signal states in a big bool array to analyze it later. This might have affected the timing as well.
I calculated the high and low counts out of this array afterwards and std::out the numbers and the bits. It turned out that sometimes the first bit was missing so I suspect the for delay might be too long sometimes:

  // Need a very short delay before reading pins or else value is sometimes still low.
  for (volatile int i = 0; i < 500; ++i) {
  }

For me it worked way better without the loop and with debounce_ok = 20. 
Have you tried several values already?

rmaes4, 2019-06-22:
@lendsle I just changed the loop as mentioned in the top comment. I saw your code but I wasn't quite sure what all I needed to change to get it working. I'm actually using a plugin for Octoprint which is like a server for my 3D printer and it references this library. I would be willing to try your code if you could give me some easy instructions as to what files I need to change and how to recompile the library.

ventilator, 2019-11-08:
I can confirm this problem with the sampe RPi3 setup as OP. Temperature readout is cut in half and humidity >100% without the fix, on every 2nd reading or so. Cable length: 20cm
[image]

ladyada, 2019-11-08:
hi this repo is going to be deprecated soon, please try 
https://github.com/adafruit/Adafruit_CircuitPython_DHT
which will replace this library

closed 2019-11-08 by ladyada