adafruit/Adafruit_Python_DHT #117: Specifiy Raspberry version when using pip3 install Adafruit_DHT

sebastianspross, 2019-09-09:
Hi,

I am trying to build a python docker container with the Adafruit_DHT library. I am wondering how I can specify the Raspberrypi version when using the `RUN pip3 install Adafruit_DHT `command.

This is my docker container:
```
FROM arm32v7/python:3-slim-stretch

RUN apt-get update \
    && apt-get install -y python3-pip \
    && python3 -m pip install --upgrade pip setuptools wheel

RUN pip3 install Adafruit_DHT

COPY MySampleAdafruitDHT.py /
WORKDIR /

CMD python3 ./MySampleAdafruitDHT.py
```

I am getting the following traceback:
``` Traceback (most recent call last):
  File "./MySampleAdafruitDHT.py", line 6, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(11, 4)
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 94, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 80, in read
    platform = get_platform()
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/common.py", line 55, in get_platform
    from . import Raspberry_Pi_2
  File "/usr/local/lib/python3.7/site-packages/Adafruit_DHT/Raspberry_Pi_2.py", line 22, in <module>
    from . import Raspberry_Pi_2_Driver as driver
ImportError: cannot import name 'Raspberry_Pi_2_Driver' from 'Adafruit_DHT' (/usr/local/lib/python3.7/site-packages/Adafruit_DHT/__init__.py)
```

Is there anything like `RUN pip3 install Adafruit_DHT --force-pi2` I can use?

ladyada, 2019-09-09:
please try https://github.com/adafruit/Adafruit_CircuitPython_DHT instead, it is agnostic

sebastianspross, 2019-09-09:
is the Adafruit_Python_DHT library deprecated in favour of the Adafruit_CircuitPython_DHT?

meanwhile I got the docker running, inspired from [farshidtz](https://github.com/farshidtz/Adafruit_Python_DHT_Docker/blob/master/Dockerfile). But I am dealing directly with the git and not pip.
```
# BUILD IMAGE
FROM arm32v6/python:3.7-alpine3.10 as builder

RUN apk --no-cache add git build-base

WORKDIR /home
RUN git clone https://github.com/adafruit/Adafruit_Python_DHT.git && \
	cd Adafruit_Python_DHT && \
	python3 setup.py install --force-pi2

## RUNTIME IMAGE
FROM arm32v6/python:3.7-alpine3.10

RUN apk --no-cache add ca-certificates

COPY --from=builder /usr/local/lib/python3.7 /usr/local/lib/python3.7
COPY --from=builder /home/Adafruit_Python_DHT/examples /home/examples

WORKDIR /home

ENTRYPOINT ["python3", "examples/AdafruitDHT.py"]
```

ladyada, 2019-09-09:
yep!

closed 2019-09-09 by sebastianspross