adafruit/Adafruit_Python_DHT #119: IOError: [Errno 24] Too many open files: '/proc/cpuinfo'

xtian777x, 2019-11-10:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry Pi 3

- Python version (run `python -version` or `python3 -version`):  Python 2.7.13

- Error message you are receiving, including any Python exception traces:  

File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT-1.4.0-py2.7-linux-armv7l.egg/Adafruit_DHT/platform_detect.py", line 9
with open('/proc/cpuinfo', 'r') as infile: 
IOError: [Errno 24] Too many open files: '/proc/cpuinfo'

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

Run a python API that has an endpoint calling Adafruit_DHT.read_retry. Make a cronjob that calls the enpoint every minute.

The problem comes from lines 65 and 92 in:

https://github.com/adafruit/Adafruit_Python_DHT/blob/master/Adafruit_DHT/platform_detect.py

The code is opening the file '/proc/cpuinfo' but not closing it. So if a Python sessions stays open, this will eventually open too many files. The solution is to close the file after reading it.

xtian777x, 2019-11-12:
Well the files are opened using with so they should be closed automatically....still not sure then why I'm getting this error...

ladyada, 2019-11-12:
we;re deprecating this library soon, please try https://github.com/adafruit/Adafruit_CircuitPython_DHT

closed 2019-11-12 by ladyada